OverseerApp/overseer #1: Unable to add a printer

pablosimone, 2018-01-09:
Hi!
I'm trying to add a printer using the windows release and I get this trace on the console.

2018-01-09 10:52:32,383 [7] ERROR Overseer.Startup.OverseerErrorHandler [(null)] - Server Error
Nancy.RequestExecutionException: Oh noes! ---> Nancy.ModelBinding.ModelBindingException: Unable to bind to type: Overseer.Core.Models.Printer ---> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.MissingMethodException: Cannot create an abstract class.
   at System.RuntimeTypeHandle.CreateInstance(RuntimeType type, Boolean publicOnly, Boolean noCheck, Boolean& canBeCached, RuntimeMethodHandleInternal& ctor, Boolean& bNeedSecurityCheck)
   at System.RuntimeType.CreateInstanceSlow(Boolean publicOnly, Boolean skipCheckThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.RuntimeType.CreateInstanceDefaultCtor(Boolean publicOnly, Boolean skipCheckThis, Boolean fillCache, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, Boolean nonPublic)
   at Nancy.Json.JavaScriptSerializer.ConvertToObject(IDictionary`2 dict, Type type)
   at Nancy.Json.JavaScriptSerializer.ConvertToType(Type type, Object obj)
   at Nancy.Json.JavaScriptSerializer.ConvertToObject(IDictionary`2 dict, Type type)
   at Nancy.Json.JavaScriptSerializer.ConvertToType(Type type, Object obj)
   at Nancy.Json.JavaScriptSerializer.ConvertToType[T](Object obj)
   at Nancy.Json.JavaScriptSerializer.Deserialize[T](String input)
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Nancy.ModelBinding.DefaultBodyDeserializers.JsonBodyDeserializer.Deserialize(String contentType, Stream bodyStream, BindingContext context)
   at Nancy.ModelBinding.DefaultBinder.DeserializeRequestBody(BindingContext context)
   at Nancy.ModelBinding.DefaultBinder.Bind(NancyContext context, Type modelType, Object instance, BindingConfig configuration, String[] blackList)
   --- End of inner exception stack trace ---
   at Nancy.ModelBinding.DefaultBinder.Bind(NancyContext context, Type modelType, Object instance, BindingConfig configuration, String[] blackList)
   at Nancy.ModelBinding.DynamicModelBinderAdapter.TryConvert(ConvertBinder binder, Object& result)
   at CallSite.Target(Closure , CallSite , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet](CallSite site, T0 arg0)
   at Nancy.ModelBinding.ModuleExtensions.Bind[TModel](INancyModule module)
   at Overseer.Modules.ConfigurationModule.<>c__DisplayClass0_0.<<-ctor>b__2>d.MoveNext() in D:\Projects\overseer\Overseer\Modules\ConfigurationModule.cs:line 18
   --- End of inner exception stack trace ---
   at Nancy.NancyEngine.InvokeOnErrorHook(NancyContext context, ErrorPipeline pipeline, Exception ex)

with this data: 
[image]

michaelfdeberry, 2018-01-10:
I have been unable to replicate this issue. Can you confirm the .Net version you have installed? 

Since this needs the a newer version you should be able to determine what version you have installed using PowerShell as outlined here: 
https://docs.microsoft.com/en-us/dotnet/framework/migration-guide/how-to-determine-which-versions-are-installed#ps_a

I will be able to test this on a different computer this evening to determine if there is any additional setup needed for windows.

michaelfdeberry, 2018-01-11:
Are you using the compiled application from the zip file on the release page or did you manually compile the code? 

I was able to successfully run the application from the zip on 2 different computers running Windows 10. I have an idea about what is failing, but I will need more information (Windows and .Net versions) before I can offer any potential solutions.

pablosimone, 2018-01-13:
Hi Michael, thank you for your time.
I'm using the zip  from the release page.
My version of .net framework is 4.7.02053 and my OS is Windows 2008R2.
[image]

michaelfdeberry, 2018-01-13:
Thanks for the additional information. 

The issue is it's using the wrong JSON serializer. Initially I thought there may have been a compatibility issue with your .Net version which was causing it not to load the Nancy.Serialization.JsonNet.dll assembly. 

However, It seems that others have had a similar issue with that assembly. https://github.com/NancyFx/Nancy.Serialization.JsonNet/issues/65

I will have to make a new release to work around this issue. Hopefully I can have that out tomorrow.

pablosimone, 2018-01-14:
it works!!! 
thanks a lot!

michaelfdeberry, 2018-01-15:
Awesome!

I am going to close this issue, but I would be interested in any feedback you may have.

closed 2018-01-15 by michaelfdeberry