OverseerApp/overseer #2: Webcam showing snow?  Bad URL conversion ..

digitalsanity, 2018-03-03:
App works great except the webcam isn't coming through quite right .. I am using cam2web on Windows which creates a stream URL as: http://192.168.0.12:8000/camera/mjpeg

This works in Octoprint, however this URL is getting mangled and is getting embedded as: http://192.168.0.12:5000http://192.168.0.12:8000/camera/mjpeg&amp;cb=1520045447144

Source URL:
<div class="webcam ng-scope" ng-if="printerCtrl.model.config.webCamUrl" ng-click="printerCtrl.toggleZoom()" style="background-image: url('http://192.168.0.12:5000http://192.168.0.12:8000/camera/mjpeg&amp;cb=1520045447144');" role="button" tabindex="0"></div>

michaelfdeberry, 2018-03-03:
Thanks for pointing this out. I will put out a new release tomorrow.

michaelfdeberry, 2018-03-03:
Version 0.0.3 should resolve this issue. If you have any other problems please create another issue.

Again, thank you for pointing this out.

closed 2018-03-03 by michaelfdeberry