OverseerApp/overseer #3: Webcam URL conversion issue

digitalsanity, 2018-03-03:
So close :)

The &cb= being added to the URL is still breaking things -- if you are adding to the querystring you need to use ? instead of & if there is not a ? already.  I'm guessing in your case your webcam stream already has a querystring and a ? in the URL, so it works to append the &cb= 

I'm not sure what the &cb= is for, but if you need it I think the only way to safely append it is to run the webcam URL through something like HttpUtility.ParseQueryString(querystring)

The ExtendQuery class here wraps ParseQueryString and handles some other edge cases nicely (like relative URLs) also and is probably the safest solution: https://stackoverflow.com/a/27545010/2538028

<div class="webcam ng-scope" ng-if="printerCtrl.model.config.webCamUrl" ng-click="printerCtrl.toggleZoom()" style="background-image: url('http://192.168.0.12:8000/camera/mjpeg**&amp;**cb=1520103941315');" role="button" tabindex="0"></div>

digitalsanity, 2018-03-03:
`<div class="webcam ng-scope" ng-if="printerCtrl.model.config.webCamUrl" ng-click="printerCtrl.toggleZoom()" style="background-image: url('http://192.168.0.12:8000/camera/mjpeg&amp;cb=1520104521158');" role="button" tabindex="0"></div>`

michaelfdeberry, 2018-03-04:
Sorry, I missed that part of it. 

The query parameter is added client side to prevent a caching issue. In this case I think the addition of the query parameter needs to stay client side. 

This should be a quick fix, but I won't be able to get to this until the morning, so look for a new release tomorrow.

michaelfdeberry, 2018-03-04:
I put out a new release, but this time I will leave this open until you confirm the issue is resolved :)

closed 2018-03-06 by michaelfdeberry