OverseerApp/overseer #5: Windows Version Doesn't Connect and Can't Close

jneilliii, 2018-06-29:
I just tried out version 0.0.4 windows release on Windows 10 and ran as administrator as given in the instructions.  When I go to the web interface and add an OctoPrint printer it is unable to connect to the printer with correct API key, etc.  I also don't seem to be able to close the application after it's started, end up having to end task in task manager.  After restarting the application the previous configurations aren't saved.

Here is the generated log in the overseer folder.

[overseer.log](https://github.com/michaelfdeberry/overseer/files/2151067/overseer.log)

jneilliii, 2018-06-29:
Should have mentioned OctoPrint info...

OctoPrint version : 1.3.9rc1
OctoPi version    : 0.15.1

michaelfdeberry, 2018-06-30:
It looks like there are some configuration changes in 1.3.9rc that currently aren't supported in 0.0.4 

The reason for this is because the current version of Overseer tries to statically map the entire config object from Octoprint API so if it changes in a newer version of Octoprint Overseer breaks. 

In the next version of Overseer changes have been made so that this is less likely to happen. I may be able to release a 0.0.5 version today. That resolves this.

> I also don't seem to be able to close the application after it's started, end up having to end task in task manager

Thanks for pointing this out. If this is something I can resolve quickly it will be included in the upcoming 0.0.5 release as well.

michaelfdeberry, 2018-06-30:
Version 0.0.5 has been released. 

I tested it with Octoprint version 1.3.9rc1 so it should resolve the configuration probably and the issue described in issue #6. 

The issue with the application not closing should be resolved as well.

If you don't mind can you confirm that the problems are resolved for you?

jneilliii, 2018-06-30:
I can confirm that 0.0.5 is working, can be closed, settings can be updated, The webcam loads for me but I don't see any details of the printer/print progress, under the name it just shows "Connecting...".  Where should print details be visible?

jneilliii, 2018-06-30:
Just realized this was with OctoPrint 1.3.8, as I just did a fresh install of OctoPi 0.15.1.  I'm going to update it to the Release Candidate and report back.

jneilliii, 2018-07-02:
Would it make a difference if I'm using the Virtual printer to test with?  I don't have any errors in the log file, just constant "Connecting..."  My live octoprint instance has additional certificate requirements for connectivity and that appears to not work at all with errors to the console.

```
System.Net.WebException: The request was aborted: Could not create SSL/TLS secure channel.
   at Overseer.Core.PrinterProviders.RestPrinterProvider.<ExecuteRequest>d__6.MoveNext() in D:\Projects\overseer_github\Overseer\Core\PrinterProviders\RestPrinterProvider.cs:line 46
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Overseer.Core.PrinterProviders.OctoprintProvider.<LoadConfiguration>d__15.MoveNext() in D:\Projects\overseer_github\Overseer\Core\PrinterProviders\OctoprintProvider.cs:line 83
2018-07-01 23:19:31,048 [6] ERROR Overseer.Startup.OverseerErrorHandler [(null)] - Server Error
Nancy.RequestExecutionException: Oh noes! ---> System.Net.WebException: The request was aborted: Could not create SSL/TLS secure channel.
   at Overseer.Core.PrinterProviders.RestPrinterProvider.<ExecuteRequest>d__6.MoveNext() in D:\Projects\overseer_github\Overseer\Core\PrinterProviders\RestPrinterProvider.cs:line 46
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Overseer.Core.PrinterProviders.OctoprintProvider.<LoadConfiguration>d__15.MoveNext() in D:\Projects\overseer_github\Overseer\Core\PrinterProviders\OctoprintProvider.cs:line 124
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Overseer.Core.ConfigurationManager.<CreatePrinter>d__8.MoveNext() in D:\Projects\overseer_github\Overseer\Core\ConfigurationManager.cs:line 52
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Overseer.Modules.ConfigurationModule.<>c__DisplayClass0_0.<<-ctor>b__2>d.MoveNext() in D:\Projects\overseer_github\Overseer\Modules\ConfigurationModule.cs:line 20
   --- End of inner exception stack trace ---
   at Nancy.NancyEngine.InvokeOnErrorHook(NancyContext context, ErrorPipeline pipeline, Exception ex)
```

jneilliii, 2018-07-02:
After adding this printer there is no way to delete the printer from the UI because when you click edit the page just shoots back over to the monitoring view of overseer with the following errors in browser console.

```
jquery.signalR.min.js:9 WebSocket connection to 'ws://localhost:9000/push/reconnect?transport=webSockets&messageId=d-1A354963-B%2C0%7CC%2C0%7CD%2C1&clientProtocol=1.5&connectionToken=AQAAANCMnd8BFdERjHoAwE%2FCl%2BsBAAAAZxZ6lXJhkEaPugEn%2Fvi%2BgAAAAAACAAAAAAAQZgAAAAEAACAAAAB4kqjA0oKZ6DETlXePgn0HKOWQBZ%2By2tflDsKlXOhUcQAAAAAOgAAAAAIAACAAAABeZpZstPnwfog%2BtrATCNcNgM0QBRiOueuYRL69RqukMDAAAACjOw3IQfpci4jmlMIbQEjXDFdPcwV%2BEcL9JPlqLkvZ%2BxaX2q8ImsPuksW%2BqQ%2F01J5AAAAAd7lIYYQDDGabNhVkBWQ3UCYzdSe0L7Bg7%2Bbn6H0b1zyBSYQ9V3cLGHzmqsmmWYJrsPC7t2OF4PoPoKKzUz87ow%3D%3D&connectionData=%5B%7B%22name%22%3A%22statushub%22%7D%5D&tid=10' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
start @ jquery.signalR.min.js:9
```

I of course figured out I can just delete the overseer.db and rebuild from scratch.

Maybe add a delete button next to edit button?

michaelfdeberry, 2018-07-02:
The virtual printer shouldn't be an issue. 

Are you running a self signed certificate? If so that is likely the problem. 

If that is the case, I would have to add support for ignoring the certificate warning to the code, but I am not sure if I want to do that. 

At least not out right, I may be willing to make it conditional so the user can enable it, or allow for the certificate to be uploaded for that instance to allow the exception for that specific certificate. 

>After adding this printer there is no way to delete the printer from the UI because when you click edit the page just shoots back over to the monitoring view of overseer with the following errors in browser console.

The delete button is part of the edit form, but the client error posted wouldn't stop the edit form from opening. I will have to do some testing to determine what happens when creating a printer and a connection to Octoprint can't be made. 

Currently it will still add the printer configuration, but I suppose this could be causing the issues your are having. I will likely change this so it won't create the printer if it can't connect to Octoprint. 

Thank you for making me aware of this. I don't use SSL/TLS with Octoprint because it's only available on my local network. I will try to look into this more and try to provide a quick solution.

jneilliii, 2018-07-02:
No rush.  Was just trying to help.  

You may want to reconsider the certificate stuff because I believe the default OctoPi image comes with https support utilizing a self-signed cert.  In my instance, I am replacing the octopi default with my own server and cert authority and user cert for access overall so if you don't have a personal certificate signed by my custom CA it won't allow access at all through haproxy.  Also, from what I can tell accessing directly via http://<ip>:5000/ appears to be blocked, which leads me to believe the OctoPi image is also pre-configured with firewall rules to only allow access to the port directly from 127.0.0.1, but I'm not 100% sure on that one.

I think you should definitely not add the printer if it can't connect, because the printer list ends up showing a row with no data on it and there is no way to delete it in that instance.

michaelfdeberry, 2018-07-03:
> No rush. Was just trying to help.

That is very much appreciated. I welcome the feedback. 

> You may want to reconsider the certificate stuff because I believe the default OctoPi image comes with https support utilizing a self-signed cert...

Right, but I wouldn't be comfortable blindly ignoring all certificate errors. It potentially allows for attack and defeats the purpose of using the https.  

I gave this some thought, I think the best way to handle this would be to prompt the user displaying the certificate details and have them confirm the exception. 

> I am replacing the octopi default with my own server and cert authority certificates and user cert for access overall so if you don't have a personal certificate signed by my custom CA it won't allow access at all through haproxy. 

It sounds like you are using client certificate authentication. I didn't get that looking at the config, but I am not that familiar with haproxy. 

That is definitely an advanced case, but it is something that could be supported pretty easily. 

>I think you should definitely not add the printer if it can't connect, because the printer list ends up showing a row with no data on it and there is no way to delete it in that instance.

That is likely what I will do, because with the way things are setup now even if the connection becomes available later the initial configuration import will never happen unless the configuration is changed again manually. 

However, it seems I may have introduced a bug because the row shouldn't be blank, and you should be able to get to the edit form.

michaelfdeberry, 2018-07-05:
I refactored the printer creation functionality and resolved the issue which didn't allow the application to be closed. 

However, I don't want to close this because there have been some other suggestions that I would like to continue to track.

jneilliii, 2018-08-31:
Thanks Michael.  I'll give it some testing and let you know how it goes.

jneilliii, 2018-09-02:
So can't seem to get the client certificate to work.  I exported my user cert as PEM format and copied it into the field and when I click save get an error "Failed to establish a connection with the printer".  Find below my log file.  I recently posted the steps for creating the SSL client connection restriction on [this](https://discourse.octoprint.org/t/restrict-access-to-octopi-utilizing-ssl-client-certificates/3681) thread post on the community forum.  I tried both the PEM export as well as a PEM chain export which gives multiple certificates in the same file (I assume the signing cert).

[overseer.log](https://github.com/michaelfdeberry/overseer/files/2342953/overseer.log)

michaelfdeberry, 2018-09-02:
Thanks for linking to the guide, that helps a lot. 

There is a null reference causing the error you are having, but fixing that alone won't make what I have in place work. 

It looks like the private key will be needed as well, which means overseer would need the password for the certificate.

What seems to work, at least on windows, is registering the certificate in the store and then only taking in the CN. Then have the application pull the cert from the store.  I should be able to do something similar on linux, but it would be a manual process to register the certificate and private key in the mono certificate store.

I don't know what the best practice is in this case, but I don't think overseer should store the private key/password. However, which would you prefer? 

1.) Providing overseer with the certificate containing the private key and the password? Or,
2.) Manually install the certificate and provide the CN to overseer?

jneilliii, 2018-09-02:
I think option 2 would be best because I already have to import the cert to use it for the browser connection. So I just need to enter the certs CN in the field for it to work?

michaelfdeberry, 2018-09-02:
> So I just need to enter the certs CN in the field for it to work?

Not with the current version, but yes that is how it will work. 

I can probably have a new version that out later today that will support this.

michaelfdeberry, 2018-09-02:
A new release is out. With the new version you just enter the common name of the certificate and it will get the certificate from the store. 

I will likely create a wiki page at some point, but for future reference.

For windows:
The certificate should be Installed for the "Current User" in the "Personal" store.

For Linux:
If the .crt and .key files are available for the client certificate concat them together into a single PEM file:
`cat client.crt client.key > client.includesprivatekey.pem`

Then use the mono certificate manager to install the PEM certificate with the command:
`certmgr -add -c  My ./client.includesprivatekey.pem`

jneilliii, 2018-09-02:
Works perfect, Now successfully able to connect with OctoPrint.  Thanks for the great work, I think this is a really cool project.

closed 2018-09-03 by michaelfdeberry