OverseerApp/overseer #9: Not able to start "exe" on rasbian

Stady234, 2018-11-08:
Following your directions, i was able to get to the end of the install where crontab was used.  however after multiple reboots, the exe does not run.  what am I missing.

I was able to get this running on a windows machine, however i could not get it to access on my hosted webpage internal to my network.

michaelfdeberry, 2018-11-08:
First confirm that mono is installed by running the `mono --version` command again. This should return something similar to this:

~~~~
Mono JIT compiler version 4.6.2 (Debian 4.6.2.7+dfsg-1)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com...
~~~~

Then try to run Overseer from a terminal using `mono /<REPLACE_WITH_PATH>/Overseer.exe` 

Please share your results. If this works I would assume the issue is with the path in the crontab configuration.

Stady234, 2018-11-08:
Ugh that case sensitive Linux!!!!  ok that worked... looks like i didn't use the "O" in the exe name.... All good now!

closed 2018-11-08 by Stady234