OverseerApp/overseer #10: Development Environment

kickhad, 2019-01-04:
Development Environment - Unclear
Windows 10, MonoDevelop 8.0.

I can compile overseer (for Windows, added a symlink to OverseerUI in the .\bin folder)
Even when run as administrator, page not found. It does make a connection. 
- tested ClientPath  = "./OverseerUI/src/" and angular icon loads. 

 	// The module cache
 	var installedModules = {};

 	// The require function
 	function __webpack_require__(moduleId) {

 		// Check if module is in cache
 		if(installedModules[moduleId]) {
 			return installedModules[moduleId].exports;
 		}
 		// Create a new module (and put it into the cache)
 		var module = installedModules[moduleId] = {
 			i: moduleId,
 			l: false,
 			exports: {}
 		};

 		// Execute the module function
 		modules[moduleId].call(module.exports, module, module.exports, __webpack_require__);

 		// Flag the module as loaded
 		module.l = true;

 		// Return the exports of the module
 		return module.exports;
 	}

 	// expose the modules object (__webpack_modules__)
 	__webpack_require__.m = modules;

 	// expose the module cache
 	__webpack_require__.c = installedModules;

 	// define getter function for harmony exports
 	__webpack_require__.d = function(exports, name, getter) {
 		if(!__webpack_require__.o(exports, name)) {
 			Object.defineProperty(exports, name, {
 				configurable: false,
 				enumerable: true,
 				get: getter
 			});
 		}
 	};

 	// getDefaultExport function for compatibility with non-harmony modules
 	__webpack_require__.n = function(module) {
 		var getter = module && module.__esModule ?
 			function getDefault() { return module['default']; } :
 			function getModuleExports() { return module; };
 		__webpack_require__.d(getter, 'a', getter);
 		return getter;
 	};

 	// Object.prototype.hasOwnProperty.call
 	__webpack_require__.o = function(object, property) { return Object.prototype.hasOwnProperty.call(object, property); };

 	// __webpack_public_path__
 	__webpack_require__.p = "";

 	// Load entry module and return exports
 	return __webpack_require__(__webpack_require__.s = 0);

// WEBPACK FOOTER //
// webpack/bootstrap 99b35345c5f7aa4bdb88

michaelfdeberry, 2019-01-04:
Overseer was developed on windows 10. The OverseerUI project was developed using Visual Studio Code. The Overseer project was developed with Visual Studio 2017.

What branch are you on? There are some major changes on the development branch that will be included in the next release. In either case you wouldn't need to create symlink. 

I have recently been updating the wiki in preparation for the next release. I will add more detailed development instructions soon, but for now here is a rough outline. This should get you going, let me know if you have any questions. 

### General
1. Ensure that node is installed and configured in your path.
2. Run the command `npm install` for the OverseerUI directory

### Master Branch

#### Development
Both project are configured to output to a bin directory in the root of the repository directory. 

1. Run `npm run watch`. This will compile the angular app and place the output in the /bin/OverseerUI directory. It will also watch for any changes in the Overseer UI so any changes made to the UI will be reflected immediately when the backend is running. 

If this fails saying the bin directory doesn't exist, either build the solution first or just create the bin directory in the root of the repository manually.

This script is defined in the package.json if you are interested where this is.

2. Run Overseer, you can either do this manually by clicking on the exe in the bin directory or running the project from visual studio with or without debugging. 

#### Release
1. Build the Overseer solution with Release configuration
2. Run 'npm run release'. This will compile the application in production mode.

The bin directory will now contain a production ready release. 

### Development Branch
The development branch is somewhat fragmented as there are 3 potential configurations. 

#### Configuration 1: .Net Core 
This is the Overseer.Core.Daemon project

The project is configured to build UI when the project is built/started from visual studio. If you set it to the startup project through visual studio it will open UI in the browser.

The project was migrated to .Net Core because it has support for Linux, Windows, and Mac natively with built in support for different architectures. 

.Net core supports many different ways to deploy, so I won't go into it here. However, I planned to release a cross platform build, that would require installing the .Net core runtime,  and a standalone linux arm build for raspberry pi. 

However, .Net core requires armv7, so it won't support lowend devices like raspberry pi zero or 1.

#### Configuration 2: .Net Framework
This is the Overseer.Mono.Daemon project

This is similar to what is currently on the master branch. The only reason this hasn't been removed is because it will work on Raspberry Pi Zero and 1 devices with mono. It's not recommended that this is used otherwise as it will likely be removed if there is no demand for supporting the low end devices.

The instructions are similar to what is outlined for the master branch. The differences are

1. The script names changed.
To watch a debug build of the project, run `npm run watchMono`
To build the project in debug mode run `npm run buildDebugMono`
To build the project in production mode run `npm run buildReleaseMono`

2. The bin directory is a child in the project directory, as typical with .Net. So running the scripts will output to the corresponding .Net project build configuration 
e.g: 
watchMono and buildDebugMono will output to Overseer.Mono.Daemon/bin/debug/OverseerUI
buildRelease will output to Overseer.Mono.Daemon/bin/release/OverseerUI

#### Configuration 3: Standalone Web App
All of the overseer functionality is implemented client side and all configuration data is stored in the browsers storage.

To build this in debug run `npm run buildDebugStatic`
To build this in production run `npm run buildReleaseStatic`

Both scripts output to the OverseerUI/dist-static. 

If you would like a preview of this, you can go to http://overseer.live.

michaelfdeberry, 2019-01-08:
> Overseer.db and AppSettings.

Could you provide some more details about this?

Did you get the error when running the code from the development branch? 

Was the database initially created by running code on the master branch, or from a release? 

If yes to both it's likely a bug in the database update process that was added in the development branch.

kickhad, 2019-01-08:
Running from the development branch, most recent commit.  It is both.

[image]

michaelfdeberry, 2019-01-09:
I have it on my todo list to write some test around the update process. However, I have run some manual test and I can't replicate the issue.

If the application threw an exception can you post the details from the log file?

kickhad, 2019-01-09:
I was able to reproduce from the development branch with Build or Release when .\overseer\src\Overseer.Core.Daemon\Overseer.db. was absent. 

The .\overseer\src\Overseer.Core.Daemon\overseer.log was empty. 

My workaround:

- comment out the valueStore steps in Program.cs and  replace "http://localhost:{appSettings.LocalPort}/" with "http://localhost:9000/"
- Do the UI Setup and shutdown
- uncomment and runs as expected. 

{
    "_id": "ApplicationSettings",
    "Value": 
    {
        "_type": "Overseer.Models.ApplicationSettings, Overseer",
        "Interval": 10000,
        "HideDisabledMachines": false,
        "HideIdleMachines": false,
        "LocalPort": 9000
    }
}
 

=== Program.cs ===
`		public static IWebHostBuilder CreateWebHostBuilder(string[] args) {			
			//var valueStore = DataContext.GetValueStore();
			//var appSettings = valueStore.Get<ApplicationSettings>();
			//var parser = new FluentCommandLineParser();
			//parser.Setup<int>("port").Callback(port => appSettings.LocalPort = port);
			//parser.Setup<int>("interval").Callback(interval => appSettings.Interval = interval);
			//parser.Parse(args);
			//valueStore.Put(appSettings);

			return WebHost.CreateDefaultBuilder(args)
				.UseStartup<Startup>()
                //.UseUrls(new[] { $"http://localhost:{appSettings.LocalPort}/" })
                .UseUrls(new[] { $"http://localhost:9000/" })
                .ConfigureLogging((hostingContext, logging) =>`