OverseerApp/overseer #11: [Request] Update Linux instructions

Stady234, 2019-01-20:
I was struggling for 30 min trying to get the overseer page to show... finally saw that the download and unpack was in the wrong folder.

I did not see at the end of  step 2 a "cd ~/" that way you are back to the /home/pi/ directory before download so that everything lines up.

Can you update the instructions... or help me figure out why it did not place in the correct directory?

michaelfdeberry, 2019-01-20:
I have been meaning to release a new version, I just haven't gotten around to it yet, but there will be updated documentation.

However, you are correct, you would need to go back to the home directory. Sorry for any confusion.

If the overseer directory is in the libsodium directory you could just move it to the home directory. Something like this should work
```
mv /home/pi/libsodium-stable/overseer /home/pi/overseer
```

michaelfdeberry, 2019-01-20:
Updated the readme

closed 2019-01-20 by michaelfdeberry