OverseerApp/overseer #12: [Issue] Can not save a user

Stady234, 2019-01-20:
When I try to add a user and click save, i get a red box that only states "An unknown error occurred".

What log and where do you need..

Application and System Information
Platform Type | Unix
Operating System | Unix 4.9.0.8
Device Name | PiVM
Application Version | 0.0.9.0

michaelfdeberry, 2019-01-20:
Going forward the self hosted version of Overseer is only recommended for people that are already familiar with deploying software. There is a web app version that is going to be recommended to everyone else. It is currently online at https://overseer.live 

If this doesn't fit your current use case I can help you get going. The log will be in the overseer directory, it's named overseer.log. 

However, the likely issue is that something went wrong with the libsodium install. Did step 2 complete without an error? 

I can't be sure until the log is posted, but if the error is something like "assembly not found" or "type not found" that is probably the case. If it is, you could try performing step to of the installation again.

michaelfdeberry, 2019-01-22:
Version 1.0.0 has been released and the libsodium install is no longer required.

Please update to version 1.0.0 or use https://overseer.live

closed 2019-01-22 by michaelfdeberry