OverseerApp/overseer #14: Overseer Daemon (Mono) - Automated

Stady234, 2019-01-28:
overseer-linux-armv 100%[===================>]   1.61K  --.-KB/s    in 0s

2019-01-28 12:09:06 (37.3 MB/s) - ‘overseer-linux-armv6.sh’ saved [1650/1650]

pi@PiVM:~ $ chmod 744 ./overseer-linux-armv6.sh && sudo ./overseer-linux-armv6.s           h
: not foundlinux-armv6.sh: 1: ./overseer-linux-armv6.sh: #!/bin/bash
: not foundlinux-armv6.sh: 2: ./overseer-linux-armv6.sh:
: not foundlinux-armv6.sh: 11: ./overseer-linux-armv6.sh:
Installing Overseer...
: not foundlinux-armv6.sh: 13: ./overseer-linux-armv6.sh:
E: Invalid operation update
: not foundlinux-armv6.sh: 16: ./overseer-linux-armv6.sh:
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package mono-complete
: not foundlinux-armv6.sh: 19: ./overseer-linux-armv6.sh:
./overseer-linux-armv6.sh: 54: ./overseer-linux-armv6.sh: Syntax error: end of file unexpected (expecting "fi")

michaelfdeberry, 2019-01-28:
What distro and what version are you running?

If you run `sudo apt-get update` manually does it complete?

michaelfdeberry, 2019-01-28:
> I am using the same Jessie Light version (but ran updates so not sure the new name) on my virtual pi that i was using for the first 0.9.0 version without issues (other than the users)

I found the problem. It has to do with the way I have git configured, when I check in it changes the line endings from unix to windows. 

> sudo apt-get update runs normal, as all other "manual" commands but with this new version i do not get the webpage, just a 503 error.

Did you setup a reverse proxy? 

If not, this is probably expected. Overseer no longer accessible outside of the host device without a reverse proxy. The reason for this is to prevent people from mistakenly making overseer accessible over the internet. 

If you have setup the reverse proxy, which did you use and can you post your configuration. 

Also, make sure the overseer service is running, you can do this by running htop and just looking for the overseer process.

Stady234, 2019-01-29:
Service is running...  On the Pi, "localhost" responds... the IP does not.  While trying to run the configuration wizard, i keep getting "An unknown error occurred" -- I have tried both the Octopi name and IP.  In a web browser on the PI the "octopi1.local" does not resolve, but the IP does.

I have tried both the nginx and haproxy using your supplied config.

No other computer on my network get anything on IP to the Pi but "Bad Request (Invalid host)" -- any combination of hostname does not resolve.

michaelfdeberry, 2019-01-29:
> Service is running... On the Pi, "localhost" responds... the IP does not

Good, the overseer service only listens on localhost, so this is expected and why a reverse proxy is needed. 

> While trying to run the configuration wizard, i keep getting "An unknown error occurred"

This sounds like an application error. If it's a backend error the details will be in the overseer.log file. Please post any relevant details.  

However, From the sounds of it you must be accessing this on the pi. I am not sure what browser is being used on the pi, but it's possible that it's not compatible with the web application. 

> No other computer on my network get anything on IP to the Pi but "Bad Request (Invalid host)" -- any combination of hostname does not resolve.

Depending on your system and router configuration the use of .local may not be needed. If the device name is "octopi1" have you tried connecting to http://octopi1/

However, lets just focus on the IP address for now. "Bad Request (Invalid host)" sounds like a reverse proxy configuration issue. 

Since the overseer service is running, and it's accessible on the device at http://localhost:9000/ this would have to be the case. 

What proxy are you currently using and did you uninstall or stop the service for the one that is not being used? If they are both trying to listen on the same port it can cause problems. 

I will review the documentation in the morning to make sure there are no errors, but could you post your configuration settings?

Stady234, 2019-01-29:
Ok, so I completely rebuild the virtual pi and stepped through slowly... I am installed and able to view local on the pi with the included chromium browser;   now for the new issues::

Pi :  Able to see overseer web on "localhost:9000" and "localhost"
       when running initial config I can add all info to printer (name/url/api) from another tab
       (so i can access printer via http://octopi1)
       but when it goes to save, i get "Unknown error occurred"  -- I have the overseer.log attached
[overseer.log](https://github.com/michaelfdeberry/overseer/files/2809149/overseer.log)

Clients on LAN:  Able to pull up the page after default nginx setup (your included config) -- initial
       After reboot of pi, no longer able to get page... "Bad Request (Invalid host)"

--
Old install work on separate virtual pi (not running at same time)

michaelfdeberry, 2019-01-29:
On one raspberry pi everything works fine, on another I was able replicate the "Bad Request (Invalid host)" issue. 

I did find a way to resolve the issue, but it requires a code change so I will have to put out a new release. I probably won't be able to get to that until later tonight.

Stady234, 2019-01-29:
ok, would it be better (since this is only for my internal use) to use nginx or haproxy?

michaelfdeberry, 2019-01-29:
I suppose that depends on your use case.  

The difference between the two is that haproxy is only a load balancer and reverse proxy, and nginx is also a web server. 

I personally use nginx, but the raspberry pi that is hosting overseer also has some other websites that need a web server. 

If you are only going to use that device/instance for overseer then I would suggest haproxy since it's a little easier to configure.

michaelfdeberry, 2019-01-30:
I was able to resolve this without a code change. 

If you are using nginx remove this line from the configuration
```
proxy_set_header        Host $host;
```

if you are using haproxy update the backend configuration to include the host header like this
```
backend overseer
        http-request set-header Host "localhost"
        server overseer1 localhost:9000
```

This seems to only be an issue with the mono version of the application. The documentation has been updated.

Stady234, 2019-01-30:
Ok, so that solved issue 1... the webpage shows on client machines after login!!
--However, I still can not add a machine as it states "an unknown error occurred"
[overseer.log](https://github.com/michaelfdeberry/overseer/files/2814263/overseer.log)

michaelfdeberry, 2019-01-30:
That is a weird error, but I was able to replicate it. 

There was a failure in my testing method. I fully tested the mono version on windows, but when testing on the device I used my existing database and just quickly walked through the application. I didn't test creating a new machine on a device. 

This seems to be specifically a mono related issue, but it's something that I am going to have to track down.

Could you provide some details about how your VM setup? If the VM supports ARMv7 maybe we can get you going with the .Net Core version while I figure out the issue.

michaelfdeberry, 2019-01-31:
This is a mono bug, but it has been fixed in a newer version. The problem is that the version in the default apt repository is out of date.

To resolve this uninstall mono and all it's dependencies 
```
sudo apt-get remove mono-complete
sudo apt autoremove
```

Add the mono repository
```
sudo apt install apt-transport-https dirmngr
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
echo "deb https://download.mono-project.com/repo/debian stable-raspbianstretch main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update
```

Install the latest version of mono
```
sudo apt-get install mono-complete
```
Verify the version 
```
mono --version
```

The current version of mono is 5.18.0.240. 

The version in the default repository, that contains the bug, is 4.6.2

Stady234, 2019-01-31:
ok, so now i can get past the setup a machine section... it says it adds.. no errors.
when i go to the main page to "monitor" nothing happens.. not even the static... just nothing.
I logged out, relogged in, deleted and resetup different printers.. no change.

michaelfdeberry, 2019-01-31:
I am not getting this error, but I am using a fresh build, I will run some test with the build from the release to determine if I can replicate the issue. 

Thanks for the help with this, and I apologize that you are having so many issues.

michaelfdeberry, 2019-02-02:
A new version, v1.0.1,  is out that should resolve this issue.

closed 2019-02-10 by michaelfdeberry