OverseerApp/overseer #17: Feature Request: Sort Machines After Adding

tophattwaffle, 2019-02-20:
The ability to sort machines in the Machine list after adding would be awesome. Right now it is "first added, first in the list" sorting.

michaelfdeberry, 2019-02-20:
I started on this today and it will likely be in the next release.

I previously had plans for a dynamic sort based on the estimated print time remaining, but I wasn't sure if there was any interest so it was scrapped.

The way I planned for that to work was adding another option in the configuration and when it was enabled the printer would be ordered from least amount of time remaining to the most, and if any machines where idle or offline the would be at the end of the list. 

If you think this is useful I will implement it as well since I am working on sorting.

tophattwaffle, 2019-02-20:
Automatic sorting based on completion would be very nice. I'm using this with ~10 machines and being able to glance over see which one is about to finish next would be awesome.

michaelfdeberry, 2019-02-24:
Reorder support has been added in 1.0.3

Sorting by time remaining is a conditional feature which can be enabled from the general tab of the configuration page. I am consider a Beta feature for now, but if you experience any issues with it please let me know.

closed 2019-02-24 by michaelfdeberry