OverseerApp/overseer #18: Can not get home page access v1.0.1 - mono - haproxy

Stady234, 2019-02-20:
I have followed the following steps per wiki instruction, but only get a "Unknown error occurred"
---

> Automated Installation
> Download
> wget https://github.com/michaelfdeberry/overseer/releases/download/1.0.1/overseer-linux-armv6.sh
> Execute
> chmod 744 ./overseer-linux-armv6.sh && sudo ./overseer-linux-armv6.sh

then

> HAProxy
> Install HAProxy
> sudo apt-get haproxy  <-- missing "install" before the work haproxy
> 
> Open the configuration file
> sudo nano /etc/haproxy/haproxy.cfg
> 
> Append the configuration below to the bottom of the file
> 
> frontend overseer-public
>         bind *:80 v4v6
>         use_backend overseer
> 
> backend overseer
>         http-request set-header Host "localhost"
>         server overseer1 localhost:9000
> 
> Restart HAProxy
> 
> sudo service haproxy restart

-----------
I can see the overseer.exe listed in htop as running...

When i open the page on my client computer (using ip) i get the main shell of page, but then the red error box in bottom right corner. "An unknown error occurred"

michaelfdeberry, 2019-02-20:
I just went through the install process on raspberry pi with a freshly installed image. I was able to get through the install and the initial setup without issue.

I suspect that your browser is likely caching the index page and it's not able to connect to the server. 

You can confirm this by clearing the cache and reloading the page. You can do this in chrome by pressing F12 to open the dev tools then right click on the reload button and click "Empty Cache and Hard Reload".

If the index page doesn't load after clearing the cache it's likely an error with the reverse proxy configuration. 

Review the configuration for potential issues. You could also take a look at the haproxy log, it's usually located at /var/log/haproxy.log

However, I suppose it could be possible that the proxy is running correctly but overseer stalled at start up leaving the process running but not starting the server. You can confirm that overseer is running and server content by curling the url from the terminal of the host device, like this:

`curl http://localhost:9000/`

You should get some html output to the console.

If the page does loads again after clearing the cache then it's likely an application error. In this case there should be some information in the log regarding the issue. There would likely be some useful information in the network and/or console tab of the dev tools as well.

Please post as much information as possible to assist with narrowing down the potential issue.

Stady234, 2019-02-26:
ok. so not sure what you updated from 1.0.1 to 1.0.3... but now with the basic install with mono/haproxy... i get the Homepage!!!

Well, I guess I was a bit to hasty in my excitement.  I can get the login page... it says i am logged in, but then nothing responds... then on a refresh i only get "503 Service Unavailable" until a reboot... once in this error state, none of the computers will connect ( main computer, multiple browsers; secondary computer, multiple browsers; pi, chromium browser)  looking for guidance... again....

michaelfdeberry, 2019-02-26:
I don't think there weren't any changes in 1.0.3 that would have resolved this issue.

We need to determine where the failure is happening before I can give you any specific direction. Please follow these steps and let me know where the point of failure occurs.

**1.) Confirm the service is running**

Type `htop` from the terminal and look for a process where the command contains `Overseer.Daemon`. If there are a lot of processes running you may need to page down. Alternatively, you can also hit F4 and type overseer to filter the list.

If the process isn't running it's either 1 of 2 things.

1. There is an error in the `/lib/systemd/system/overseer.service` file. Here is an example of a working service configuration.

```
[Unit]
Description=Overseer Daemon

[Service]
WorkingDirectory=/home/pi/overseer
ExecStart=/usr/bin/mono /home/pi/overseer/Overseer.Daemon.exe
ExecReload=/bin/kill -HUP
KillSignal=SIGINT

[Install]
WantedBy=multi-user.target
```
Please compare this to what you have and make any needed changes. Then run `sudo service overseer start`. You may get a message saying you need to run another command since the configuration changed, if so just run the specified command

I believe the command is ` sudo sudo systemctl daemon-reload`, in any case it will be in the error message if you need to perform this action.

2. Overseer failed at startup.

In this case the error should be in the overseer.log in the overseer directory. However, I suppose it's possible it could fail before the logging is initialized. 

You will want to start overseer from the terminal to get the error message. Do this by changing into the overseer directory, with the default install that would be `cd ./overseer`. Then execute the command `mono Overseer.Daemon.exe`. If it fails the error will be logged to the terminal.

If overseer starts without error you will see something like  "listing at localhost:9000" the service started correctly. If this is the case hit Ctrl+C to kill the process and continue to step 3. 

**2.) Confirm that overseer is serving content**

Confirm content is being served by accessing http://localhost:9000 from the device either using a browser on the device or using curl from the terminal. 

This is similar to case 2 of item 1. The error should be in the overseer.log file. In the case that there are no errors in the log file run the command `sudo service overseer stop` and then follow the steps above to start overseer from the terminal.

**3.) Confirm the haproxy configuration is correct**

If the service is running and serving content then the issue is with the haproxy configuration. The important part of the haproxy configuration, compared to the nginx, is that the configuration should be appended to the file and not replaced. Here is a full example of a working config file.

```
global
	log /dev/log	local0
	log /dev/log	local1 notice
	chroot /var/lib/haproxy
	stats socket /run/haproxy/admin.sock mode 660 level admin
	stats timeout 30s
	user haproxy
	group haproxy
	daemon

	# Default SSL material locations
	ca-base /etc/ssl/certs
	crt-base /etc/ssl/private

	# Default ciphers to use on SSL-enabled listening sockets.
	# For more information, see ciphers(1SSL). This list is from:
	#  https://hynek.me/articles/hardening-your-web-servers-ssl-ciphers/
	# An alternative list with additional directives can be obtained from
	#  https://mozilla.github.io/server-side-tls/ssl-config-generator/?server=haproxy
	ssl-default-bind-ciphers ECDH+AESGCM:DH+AESGCM:ECDH+AES256:DH+AES256:ECDH+AES128:DH+AES:RSA+AESGCM:RSA+AES:!aNULL:!MD5:!DSS
	ssl-default-bind-options no-sslv3

defaults
	log	global
	mode	http
	option	httplog
	option	dontlognull
        timeout connect 5000
        timeout client  50000
        timeout server  50000
	errorfile 400 /etc/haproxy/errors/400.http
	errorfile 403 /etc/haproxy/errors/403.http
	errorfile 408 /etc/haproxy/errors/408.http
	errorfile 500 /etc/haproxy/errors/500.http
	errorfile 502 /etc/haproxy/errors/502.http
	errorfile 503 /etc/haproxy/errors/503.http
	errorfile 504 /etc/haproxy/errors/504.http

frontend overseer-public
	bind *:80 v4v6
	use_backend overseer

backend overseer
	http-request set-header Host "localhost"
	server overseer1 localhost:9000

```

Stady234, 2019-02-26:
[overseer.log](https://github.com/michaelfdeberry/overseer/files/2907694/overseer.log)
Attached is the log file...
--------------
So While checking i found the following:
Service was running on first connect.. opened webpage from client computer.. on ligin, service crashed (no longer listed in htop)
restarted service and refreshed webpage... now presented with "add machine".. after filling out, and clicking save, I only receive "an unknown error"

michaelfdeberry, 2019-02-26:
I think this is the same mono bug that was discussed here https://github.com/michaelfdeberry/overseer/issues/14#issuecomment-459374723. 

Please run the `mono --version` command and post what version you have installed.

I did update the install script to install the latest version of mono from the correct repository, but if this is the case the likely cause here is that the older version of mono was already installed so the install command  in the script didn't do anything.

Please follow the steps outline in the linked comment and if that resolves the issue I will update the install script to check if mono exist and if it does perform an update instead of trying to install.

Stady234, 2019-02-27:
Briliant!!  I am sorry for the trouble...  I forgot that when i was building this virtual pi, it was Debian based, not raspian... so the base code you have is limited to the 4.6.2.7+dfsg-1... i had to use the following to get it to work on this "variation" environment...
```
sudo apt install apt-transport-https dirmngr gnupg ca-certificates
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
echo "deb https://download.mono-project.com/repo/debian stable-stretch main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update
```

everything else is working as intended from what i see... I will make sure i update my notes so to not forget my Virtual Pi variation.

michaelfdeberry, 2019-02-27:
It's no trouble, at some point I will migrate these comments to the wiki as a troubleshooting guide.

closed 2019-02-27 by michaelfdeberry