OverseerApp/overseer #19: Printer Display Size Incorrect When Machines Are Hidden

tophattwaffle, 2019-02-22:
With hide idle machines, and hide disabled machines enabled, with a single printer active, the camera will only take up a small portion of the window unless focused by clicking on it.

[image]

[image]

With both of these settings off, the window displays all printers as expected.
[image]

michaelfdeberry, 2019-02-22:
I pushed a fix to the dev branch that will be included in the next release. That will probably be out this weekend when the sorting functionality is finished.

I guess a temporary fix would be resizing the browser window as that will trigger the code to resize the grid.

tophattwaffle, 2019-02-22:
You're doing the lords work. Resizing works great for now.

michaelfdeberry, 2019-02-24:
Resolved in 1.0.3

closed 2019-02-24 by michaelfdeberry