OverseerApp/overseer #20: Feature Request: User Permissions

tophattwaffle, 2019-03-21:
My goal is to set up a kiosk / display that will show Overseer, however I don't want any user on this computer to have access to the printer controls.

Essentially an option to make a user "View only" where only webcams are shown and not the controls / overseer options.

michaelfdeberry, 2019-03-22:
Good idea.

michaelfdeberry, 2019-03-26:
This functionality was added with the 1.0.4 release. For now there are only 2 types of users readonly and administrators. 

I was thinking about more granular permission controller and allowing for custom profiles, however that may be overkill for this application. However, I may add support for a restricted user at some point

If you are using the standalone client build, or overseer.live, please take note of the warning on the release page. If the user has the ability to open the the browser's dev tools they will have access to the database, so please express caution if when using that functionality. The client/server builds don't have this issue since the database is on a remote device.

closed 2019-03-26 by michaelfdeberry

tophattwaffle, 2019-03-26:
Man you added that quick! Awesome!

Just tested by upgrading my instance. I created a new user (Guest) and when logged in, I cannot make changes. 

When viewing the user screen, there isn't a column for what user has what role. Really not a huge deal, but would be nice to see.

[image]

michaelfdeberry, 2019-03-26:
Yeah, it should be there. Thanks for pointing that out.