OverseerApp/overseer #21: [Question] Browser Memory usage

Stady234, 2019-04-08:
First off... this version (1.0.8) is working like a champ!!! I am loving it!
- I would suggest to find a way that when you run the first "auto install" script that it auto pulls from your latest version.. as is (per wiki) i had to mod the git line to reflect the 1.0.1 -> 1.0.8...
===
ok on to the real question.... I am trying to narrow down if it is just chrome acting weird.. but when i am displaying the Overseer (virtual pi mono install) my system takes a huge performance hit...
I have an i7, 32GB ram.. so i dont believe it is a physical limitation, but software.  I am currently under the impression it is either chrome or Win7 causing it.  But i figured i would cover all basis and ask the creator if  anything else to look into.

==
Thanks again for this newest update!

michaelfdeberry, 2019-04-09:
It's possible there could be a memory leak client side, I will look into it. 

In the meantime can you use the Chrome Task Manager to give me a rough idea of the CPU and Memory utilization for the client app? You can open it by pressing Shift+Esc, or right clicking on the top of the window frame and clicking Task Manager. 

The wiki is a different repository, but I may be able to update the build script to check out and update the wiki pages. I will add it to the TODO list, but it's going to be low priority.

michaelfdeberry, 2019-04-10:
I haven't had a lot of time to look at this yet, so I don't have anything conclusive. However, from what I am seeing the tab doesn't use much memory, but the GPU process memory usage is much higher when Overseer is open. That is likely from the processing of the mjpeg stream. 

IIRC you have, or at least close to, 10 printers configured. If they are all high-ish res with high-ish frame rates I suppose you could run into this issue.

Again, nothing conclusive at this point, but I will keep looking. I will also attempt to do some testing with multiple streams as I haven't tested with more than 4.

If you could follow up with the information from the previous comment that would be helpful. If some of your printers are idle you could enable "Hide Idle Printers" and see if there is a difference vs having them all displayed.

Stady234, 2019-04-14:
Looks like Microsoft is messing with me... since it is a Win7 machine, they are sending weird updates in prep for "forcing" the win10 upgrade.  After forcing a few updates, especially the video driver, system is running much better.  Please disregard as this was my only my machine!

closed 2019-04-14 by Stady234