OverseerApp/overseer #22: [Question] No login mode for live-stream rebroadcast?

Stady234, 2019-04-14:
Just wondering if you already have (and if not maybe as a request) to have access with no login if only to display during a livestream.
--
Background::  During my live stream, I frequently like to display all the printers in my StreamLabs OBS in a "browser source" .  0.9 was able to do this... without a login.  The new version, which i would prefer to use, but it stops at a login that the "source" does not allow for keyboard input.

I do not allow outside my network access to the current Overseer host.  Which is hosted via VirtualPi on my desktop.

If i need to create another virtual to host a different version just for this I can, but that is why i figured I would ask your guidance.

michaelfdeberry, 2019-04-14:
I removed all support for anonymous access, so there is currently no way to do this, and I am reluctant to add back such functionality. 

I might be able to make it little easier to access a readonly user account without a manual login. My initial thought is to allow the generation of one time use token that can be applied to the url to perform the login.

However, let me think about this a little bit to make sure this doesn't compromise security otherwise.

Stady234, 2019-04-15:
No problem... like i said as i am only showing the video feed in the stream, if it was a read only, video feed only... would be awesome... but by no means go out of your way with too much work/investment time in this... I do have a workaround, so this is no rush.

Either way, thank you in advance for your effort.

michaelfdeberry, 2019-06-11:
Just an update. I added this functionality, but I have some more testing to perform. 

The next release will likely be out this coming weekend.

michaelfdeberry, 2019-06-19:
This functionality is now available in version 1.0.13. 

When logged in as an Admin user the option to generate a URL will appear on the edit page for read-only users.

closed 2019-06-19 by michaelfdeberry