OverseerApp/overseer #24: Windows 7 Behavior

DavidWhelan, 2019-06-10:
I don't know if you are aware of this issue or not but it appears computers running Windows 7 will not display the camera feed for Octoprint. It will connect to the printer and display bed/hotend temperatures as well as the start/stop controls. It will show the video feed on a Windows 10 machine.

The instance of Overseer is running on one of the Octoprint servers.

michaelfdeberry, 2019-06-10:
Does the camera feed display from the Octoprint UI?
Does the camera feed work if the stream URL is opened in the browser?
What browser are you using?

DavidWhelan, 2019-06-11:
The Camera feed does display on the Octoprint UI and the camera feed is working from the webcam URL when opened in browser. I tested both of these things on the machine in question. I am currently using Google Chrome to test it.

michaelfdeberry, 2019-06-11:
If it can open in Octoprint and in the browser it should work for Overseer.

Confirm that the url in the printer configuration is correct. 

If the url is correct, browse to Overseer in Chrome and open the dev tools, by pressing F12, then refresh the page. If the camera feed is failing to load there should be errors in the console tab, please copy and paste them here.

DavidWhelan, 2019-06-12:
I tracked down the issue. The printer configuration used hostname.local as the URL to access the octoprint instances. The raspberry pi computer didn't have any problems connecting to this url to gather and display the connection info. However, the Windows 7 machine didn't have the Bonjour print driver installed so it couldn't resolve hostname.local. We have been using hostname/ to access the printers so it has never been an issue. The Windows 7 machine could get the connection info since it was the pi that retrieved and generated the info on the page, but the videos wouldn't show up because they were just containers for the video link which uses the hostname.local which wouldn't resolve on the machine. I changed the video links in the configuration to use http:hostname/ instead of the hsotname.local and the video came right up.

closed 2019-06-13 by DavidWhelan