OverseerApp/overseer #25: [Request] Preauthenticated Url - Time extension

Stady234, 2019-06-24:
Is there any way to make the valid time of the url be admin configurable?  I use during my 3hr live-stream and would like it to work the entire time.  I have changed the read-only user on all time options.. 1day through 90.. and after two min, I can no longer use the page link in my OBS.  It is currently set to use a browser window, and i pasted the "custom" url in the object config.  It works initially and I can swap to my different scenes and back just fine.  However, after the two min.. I can no longer swap to the scene with the overseer link.

michaelfdeberry, 2019-06-24:
I am guessing the session is still active, but that it's reloading the login page. 

Are you able to interact with the webpage? If so can you click on "Overseer" in the top left corner. If it's goes the homepage then the session is still active. 

If that is the case what I can do is update the login page to redirect if there is an active session.

Stady234, 2019-06-25:
when i pull it into OBS as a browser source... no, there is no webpage direct interaction... I will try again with that pre-auth page in a "private" browser session on the main pc... that way I can check what the behavior is doing...

michaelfdeberry, 2019-07-14:
Any updates on this? I suppose what I really need to know is if any session information is stored. For any other website is any state retained when the page is reloaded?

If you are unsure I can attempt to replicate your setup to perform some test. However, I know nothing about OBS. If it's fairly simple to setup could you provide a brief overview of your setup. If it's more complex can you provide links to help get me going.

Stady234, 2019-07-14:
ok, so the way it has been working... My OBS loads a web page as if it was the first time "activating" a page.. every time.. so no persistence info is stored..  Since i am calling it as a "browser source (default setup) it will not allow me to mouse click on it.  If i used a "capture desktop" then opened the page, everything works since i am using a standard interface without OBS.  This however defeats the purpose.  Let me know if you need me to record a session and send you the link to see the problem...

michaelfdeberry, 2019-08-11:
When you get a chance try 1.0.15. I finally had time to look into how the OBS BrowserSource works, hopefully the latest changes resolves your issue.

Stady234, 2019-08-12:
New version is working GREAT!!!! I have run an hour swap scene test (just like during my livestream) and so far no issues what so ever!!!
--
2 Side questions  (if this needs its own thread) - 
1. What is the "profile" on the machine page?
2. The "about" page shows a version behind.. is that normal? Not an issue for me as this is working, but for issue tracking...
[image]

michaelfdeberry, 2019-08-12:
1.) That is the Octoprint profile for the printer. It's there mostly for debugging just to confirm that Octoprint and Overseer are using the same profile. Thanks for pointing that out, it's actually supposed to be readonly. I will also change the label to "Octoprint Profile", to reduce confusion. 

2.) I checked the assembly and it does indeed have the wrong version. This is a strange one, I will have to look into this more.

However, this is probably a failure of the build script I created. There were no backend changes for this version so it probably just reused the existing files. I need to ensure the old files are deleted before the build runs. 

That script was just something I cobbled together so I wouldn't have manually package everything up so it's probably not as robust as it needs to be. 

I don't need a new issue for either of these, but thank you for bringing it to my attentions!

Stady234, 2019-08-12:
Thank you for the update... and now that you mention it.. the "profile" name did match octoprint profile... not sure why i didn't make that connection.. but either way.. looks like everything is working great!

As i don't have any more issues... i will close this ticket!  Please feel free to msg me anytime if you need more testing!

closed 2019-08-12 by Stady234