OverseerApp/overseer #26: Octoprint always "offline"

Sebclem, 2019-07-30:
When i connect a printer, it always say that the printer is offline.
I have somme error on the js console :
[image]

and that the error line:
[image]

I don't have any error on network and the json seems good:
[image]

Sebclem, 2019-07-30:
I have noticed that it's only when i launch a print, if the printer is "idle" it's ok
[image]

michaelfdeberry, 2019-07-30:
Can you provide a few more details about your setup? 

Since there are a couple of different variants I need to know which setup you are using.

If you are using one of the hosted backend solutions (.Net Core or Mono), then please post the overseer.log from the overseer directory.

If you are using the standalone webapp solution (Overseer.live or standalone webapp), please expand the error message copy the entire stack trace and post it in a comment. 

It looks like you are using the webapp version, but looking at the Octoprint response is fine. I run some test this evening when I get home.

Sebclem, 2019-07-30:
I'm using the webapp version (http://overseer.live/) with Octoprint 1.3.11.
is that the stack-trace you want ? This is the only logs i have.
```main.130688542289751c786f.js:1 ERROR unknown_exception
$r @ main.130688542289751c786f.js:1
t.handleError @ main.130688542289751c786f.js:1
next @ main.130688542289751c786f.js:1
o @ main.130688542289751c786f.js:1
e.__tryOrUnsub @ main.130688542289751c786f.js:1
e.next @ main.130688542289751c786f.js:1
e._next @ main.130688542289751c786f.js:1
e.next @ main.130688542289751c786f.js:1
e.next @ main.130688542289751c786f.js:1
e.emit @ main.130688542289751c786f.js:1
(anonymous) @ main.130688542289751c786f.js:1
e.invoke @ polyfills.7c27b50cded1d38e7ae3.js:1
t.run @ polyfills.7c27b50cded1d38e7ae3.js:1
t.runOutsideAngular @ main.130688542289751c786f.js:1
onHandleError @ main.130688542289751c786f.js:1
e.handleError @ polyfills.7c27b50cded1d38e7ae3.js:1
t.runTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.invokeTask @ polyfills.7c27b50cded1d38e7ae3.js:1
invoke @ polyfills.7c27b50cded1d38e7ae3.js:1
n.args.<computed> @ polyfills.7c27b50cded1d38e7ae3.js:1
setTimeout (async)
s @ polyfills.7c27b50cded1d38e7ae3.js:1
e.scheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
onScheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
e.scheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.scheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.scheduleMacroTask @ polyfills.7c27b50cded1d38e7ae3.js:1
d @ polyfills.7c27b50cded1d38e7ae3.js:1
(anonymous) @ polyfills.7c27b50cded1d38e7ae3.js:1
M.a.<computed> @ polyfills.7c27b50cded1d38e7ae3.js:1
r @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e.notifyError @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e.notifyError @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
(anonymous) @ main.130688542289751c786f.js:1
t._trySubscribe @ main.130688542289751c786f.js:1
t.subscribe @ main.130688542289751c786f.js:1
a @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
e._error @ main.130688542289751c786f.js:1
e.error @ main.130688542289751c786f.js:1
l @ main.130688542289751c786f.js:1
e.invokeTask @ polyfills.7c27b50cded1d38e7ae3.js:1
onInvokeTask @ main.130688542289751c786f.js:1
e.invokeTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.runTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.invokeTask @ polyfills.7c27b50cded1d38e7ae3.js:1
m @ polyfills.7c27b50cded1d38e7ae3.js:1
k @ polyfills.7c27b50cded1d38e7ae3.js:1
[... 1004 lines cut ...]
onScheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
e.scheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.scheduleTask @ polyfills.7c27b50cded1d38e7ae3.js:1
t.scheduleMicroTask @ polyfills.7c27b50cded1d38e7ae3.js:1
I @ polyfills.7c27b50cded1d38e7ae3.js:1
t.then @ polyfills.7c27b50cded1d38e7ae3.js:1
t.appInitializer @ main.130688542289751c786f.js:1
t.runInitializers @ main.130688542289751c786f.js:1
(anonymous) @ main.130688542289751c786f.js:1
(anonymous) @ main.130688542289751c786f.js:1
e.invoke @ polyfills.7c27b50cded1d38e7ae3.js:1
onInvoke @ main.130688542289751c786f.js:1
e.invoke @ polyfills.7c27b50cded1d38e7ae3.js:1
t.run @ polyfills.7c27b50cded1d38e7ae3.js:1
t.run @ main.130688542289751c786f.js:1
t.bootstrapModuleFactory @ main.130688542289751c786f.js:1
zUnb @ main.130688542289751c786f.js:1
i @ runtime.44938c253f7c297068a8.js:1
0 @ main.130688542289751c786f.js:1
i @ runtime.44938c253f7c297068a8.js:1``
r @ runtime.44938c253f7c297068a8.js:1
t @ runtime.44938c253f7c297068a8.js:1
(anonymous) @ main.130688542289751c786f.js:1
Show 97 more frames
```

michaelfdeberry, 2019-07-31:
I removed the stack trace because it was too big :) It also didn't provide any value, so I need to improve logging for the client side app. 

I haven't been able to replicate the issue. My printer/pi using Octoprint 1.3.11 is working fine with overseer.live. 

The response json you posted looks fine, so that shouldn't be the issue. The only other thing it does when creating the job status is access the tools configuration for the machine.

If you open the dev tools and look under Application > IndexedDB > machines, you should see something like this.

[image]

Can you expand the tools collection and take a similar screenshot? 

Also, could you add the same printer again in Overseer, but with a different name, and see if the second setup has the same issue?

Sebclem, 2019-07-31:
I found what cause the problem, i have 2 extruder, if i put only one extruder on my profile on Octoprint, that's working.
I thought that was because i have a "shared" nozzle but i sill have the issue if i un-check it, that's my config on octoprint: 
[image]

This is the screenshot you asked:
[image]

michaelfdeberry, 2019-07-31:
I have confirmed that is the cause and was able to replicate the issue.

I will try to put out  a new release tonight, however because of how overseer.live is setup it might take up to 24 hours before you actually see the newer version. I will post again when the update is released.

Thanks for bringing this to my attention!

michaelfdeberry, 2019-07-31:
I have committed a fix and pushed an update to overseer.live. As stated this could take up to 24 hours, however I am already seen version 1.0.14 on overseer.live.

I am going to close this, but please let me now if you have any other problems. Thanks!

closed 2019-07-31 by michaelfdeberry

Sebclem, 2019-07-31:
Thank you! You're app is awesome, keep going !