OverseerApp/overseer #27: Overseer will not connect after enabling SSL

gmccauley, 2019-09-05:
I'm running Overseer 1.0.15 .NET Core in RPi 3B+

I've got two octoprint (octipi image) instances, and when I add them to overseer, I see the webcam image, but it's stuck "Connecting"

[image]

The overseer.log is repeating the following:
`2019-09-05 17:14:17,059 [70] ERROR Microsoft.AspNetCore.SignalR.HubConnectionContext [(null)] - Failed connection handshake.`

I noticed that this started after adding SSL settings to the nginx config.

```
server {
        listen 9001 default_server;
        ssl_certificate                 /etc/nginx/cert/overseer.cer;
        ssl_certificate_key             /etc/nginx/cert/overseer.key;
        ssl                             on;
        ssl_session_cache               builtin:1000  shared:SSL:10m;
        ssl_protocols                   TLSv1.1  TLSv1.2;
        ssl_ciphers                     ECDH+AESGCM:ECDH+AES256:ECDH+AES128:DH+3DES:!ADH:!AECDH:!MD5;
        ssl_prefer_server_ciphers       on;

        access_log                      /var/log/nginx/nginx.log;

        location / {
                proxy_pass              http://localhost:9000;
                proxy_http_version      1.1;
                proxy_cache_bypass      $http_upgrade;
                proxy_set_header        Upgrade $http_upgrade;
                proxy_set_header        Connection keep-alive;
                proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header        X-Forwarded-Proto $scheme;
                proxy_set_header        Host $host;
        }
}
```

If I turn ssl to off, it works as expected.

michaelfdeberry, 2019-09-05:
I may have to look into this more, but try replacing the line: 
`proxy_set_header Connection keep-alive;`

with 
`proxy_set_header Connection "upgrade";`

If this fails with a different error please post the error message

gmccauley, 2019-09-06:
> I may have to look into this more, but try replacing the line:
> `proxy_set_header Connection keep-alive;`
> 
> with
> `proxy_set_header Connection "upgrade";`
> 
> If this fails with a different error please post the error message

So that worked, however if I logout, I'm unable to log back in.  No errors in the log.

michaelfdeberry, 2019-09-06:
That's interesting. 

If the error isn't in the log what is probably happening is that the connection isn't making it to overseer. If that is the case there should be something in console under the dev tools for the browser. You can open that by pressing F12 and then attempting to login. If an error is logged there please post it here.

I have not tested this at all, but what might need to happen is to setup a different location for the websocket connection. 

To do that revert your nginx config to use `proxy_set_header Connection keep-alive;` and try add this additional location:

```
location /push {
    proxy_pass              http://localhost:9000;
    proxy_http_version      1.1;
    proxy_cache_bypass      $http_upgrade;
    proxy_set_header        Upgrade $http_upgrade;
    proxy_set_header        Connection "upgrade";
    proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header        X-Forwarded-Proto $scheme;
    proxy_set_header        Host $host;
}
```
What this does is use the initial configuration you had for API calls (login, configuration, etc ), and uses different configuration for the web socket (printer status updates).

if it has trouble connecting you might need to change the proxy_pass to "http://localhost:9000/push" 

Otherwise, I should have some time this weekend to configure nginx with tls and replicate the issue to give you a more confident solution.

gmccauley, 2019-09-06:
Thank you so much for the quick and helpful responses.  It's working now!!

Here's my working nginx config for reference.
```
server {
        listen 9001 default_server;
        ssl_certificate                 /etc/nginx/cert/overseer.cer;
        ssl_certificate_key             /etc/nginx/cert/overseer.key;
        ssl                             on;
        ssl_session_cache               builtin:1000  shared:SSL:10m;
        ssl_protocols                   TLSv1.1  TLSv1.2;
        ssl_ciphers                     ECDH+AESGCM:ECDH+AES256:ECDH+AES128:DH+3DES:!ADH:!AECDH:!MD5;
        ssl_prefer_server_ciphers       on;

        access_log                      /var/log/nginx/nginx.log;

        location / {
                proxy_pass              http://localhost:9000;
                proxy_http_version      1.1;
                proxy_cache_bypass      $http_upgrade;
                proxy_set_header        Upgrade $http_upgrade;
                proxy_set_header        Connection keep-alive;
                proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header        X-Forwarded-Proto $scheme;
                proxy_set_header        Host $host;
        }

        location /push {
                proxy_pass              http://localhost:9000/push;
                proxy_http_version      1.1;
                proxy_cache_bypass      $http_upgrade;
                proxy_set_header        Upgrade $http_upgrade;
                proxy_set_header        Connection "upgrade";
                proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header        X-Forwarded-Proto $scheme;
                proxy_set_header        Host $host;
        }
}
```

closed 2019-09-06 by gmccauley