OverseerApp/overseer #29: Unable to add printers in Edge 78.0.276.20 (overseer.live)

Crowlord, 2019-11-01:
When attempting to add machines it cannot connect to machine. 

Connection with same parameters works fine in Firefox.

michaelfdeberry, 2019-11-01:
Are you getting any error messages? Either via popup notification or in the developer tool console? 

That is the preview of the Chromium based version of Edge correct? 

I will take a look at it this weekend, and if the issue is with the app I will fix it. However, if it's a browser specific issue I probably won't add a work around for prerelease software.

Crowlord, 2019-11-01:
Right I am going to the Pub before I destroy all technology. 

Its literally just worked fine. I haven't changed anything so tech clearly hates me again.

The third printer wont connect but I think that may be because I dont have CORS on yet on it.

closed 2019-11-01 by Crowlord

michaelfdeberry, 2019-11-01:
It happens lol. I'll still take a look this weekend. I've been meaning to beef up the logging for the client build and make it easier to share logs in general.

This could have been as simple as the connection to the printer timing out or failing.