OverseerApp/overseer #30: [Feature Request]

taker218, 2019-11-15:


michaelfdeberry, 2019-11-15:
Yes, I will make sure it's in the next release. 

I don't know how much time I will have to work on Overseer, but I hope to have a new version out either this weekend or the next.

taker218, 2019-12-02:
Any news on the next release?

Or is there a way I could test the changes you made? I saw that you made some changes in the code regarding this topic.

michaelfdeberry, 2019-12-02:
I discovered a few bugs that delayed the release, but I can potentially put a release out this evening

michaelfdeberry, 2019-12-02:
I put out a release that adds this functionality. For now it just supports flipping the image, but I might add support for rotating at some point.

taker218, 2019-12-02:
awesome, thanks for the release, will test it right away

michaelfdeberry, 2019-12-02:
If you use overseer.live the changes my not be available immediately. 

I am going to close this, but if you find any problems please open a new issue.

closed 2019-12-02 by michaelfdeberry