OverseerApp/overseer #31: Unknown error message in log

taker218, 2019-12-12:
So, I just used the new version for quite a while now and just realised, that there are a lot of unknown errors in the log.

I use it on a Raspberry Pi 4 and so far it's working well. But everytime I connect to the webinterface via reverse proxy, so I can use it on my network, I get an unknown error message in the bottom right.

Here are the last lines of the overseer.log
[overseer_20191212.log](https://github.com/michaelfdeberry/overseer/files/3956410/overseer_20191212.log)

michaelfdeberry, 2019-12-12:
I added some code to log the underlying error message and not the key, the key is just used for the alert text, but I must have inversed that logic by mistake. 

So something is failing, but the wrong message is being logged.

You should be able to get the actual error using the browsers dev tools. If you can please post it here.

I don't think this is related to the pi 4, but I could run some test. Could you provide details about your setup?

taker218, 2019-12-12:
Will do so when I'm at home and on my network.

taker218, 2019-12-13:
I just had a look at the console in firefox and got the following errors.
Hope this helps, if you need more, just let me know.

[overseer_firefox_log.txt](https://github.com/michaelfdeberry/overseer/files/3961077/overseer_firefox_log.txt)

michaelfdeberry, 2019-12-13:
There is an issue with the logging. In this case the unknown error popup should be displayed, but the correct message should be in the log. I should be able to spend some time on that this weekend. 

As for why you are getting the error, it's likely related to issue #27

Using SSL/TLS with nginx requires a little more configuration. You can resolved this by configuring a separate location for the websocket. See the example below. 

If you aren't using nginx, it's still likely a reverse proxy issue. Please post the configuration for the proxy server you are using and I will assist getting it sorted. 

I will add this to the wiki, but here is an example config for using nginx with SSL/TLS
```
server {
        listen 9001 default_server;
        ssl_certificate                 /etc/nginx/cert/overseer.cer;
        ssl_certificate_key             /etc/nginx/cert/overseer.key;
        ssl                             on;
        ssl_session_cache               builtin:1000  shared:SSL:10m;
        ssl_protocols                   TLSv1.1  TLSv1.2;
        ssl_ciphers                     ECDH+AESGCM:ECDH+AES256:ECDH+AES128:DH+3DES:!ADH:!AECDH:!MD5;
        ssl_prefer_server_ciphers       on;

        access_log                      /var/log/nginx/nginx.log;

        location / {
                proxy_pass              http://localhost:9000;
                proxy_http_version      1.1;
                proxy_cache_bypass      $http_upgrade;
                proxy_set_header        Upgrade $http_upgrade;
                proxy_set_header        Connection keep-alive;
                proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header        X-Forwarded-Proto $scheme;
                proxy_set_header        Host $host;
        }

        location /push {
                proxy_pass              http://localhost:9000/push;
                proxy_http_version      1.1;
                proxy_cache_bypass      $http_upgrade;
                proxy_set_header        Upgrade $http_upgrade;
                proxy_set_header        Connection "upgrade";
                proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header        X-Forwarded-Proto $scheme;
                proxy_set_header        Host $host;
        }
}
```

taker218, 2019-12-13:
I'm using apache as a reverse proxy and this is the corresponding entry in the config.
```
<VirtualHost *:8000>
        ProxyPreserveHost On
        ProxyPass / http://localhost:9000/
        ProxyPassReverse / http://localhost:9000/
        ProxyPass /push http://localhost:9000/push
        ProxyPassReverse /push http://localhost:9000/push

        RequestHeader set X-SCHEME https

        Timeout 5400
        ProxyTimeout 5400

</VirtualHost>
```

The bit with /push was just added by me just now. It definitely connects faster to the printers now, but the error message in the console is still there.

michaelfdeberry, 2019-12-13:
I am not familiar with Apache, but this is what I have found so far.

Apache won't change protocols automatically, so you have to add a rewrite to handle the websocket request.

```
<VirtualHost *:8000>
        RewriteEngine on
        RewriteCond ${HTTP:Upgrade} websocket [NC]
        RewriteCond ${HTTP:Connection} upgrade [NC]
        RewriteRule .* "ws:/localhost:9000/$1" [P,L]
        
        ProxyPreserveHost On
        ProxyPass / http://localhost:9000/
        ProxyPassReverse / http://localhost:9000/
        RequestHeader set X-SCHEME https
        Timeout 5400
        ProxyTimeout 5400

</VirtualHost>
```

You will also need to make sure websocket tunneling is enabled, do that by executing the following command. 
```
a2enmod proxy_wstunnel
```

It also mentions the need for the following commands, but since the proxy is working otherwise I am assuming this may have been done.
```
a2enmod proxy
a2enmod proxy_http
```

taker218, 2019-12-13:
It didn't really worked with the apache reverse proxy and it seemed like it was a problem with my setup here.

I installed nginx now and used the config from your earlier post.

Used a different ssl_cert (since I have one from let's encrypt. And it worked when I connected to the Pi directly via url.
I also have a Synology Diskstation in between the internet and my Pi, so I have one certificate for everything (VPN, FTP, HTTPS). This was kind of a culprit there.

Will now write myself a script, which will copy the right certificate onto my pi every night.

Thanks for your help and for looking into it with me!

michaelfdeberry, 2019-12-13:
If Apache works better for you I have no problem assisting you getting it to work. However, I won't be able to actually install Apache to test any configuration until maybe later tonight or tomorrow morning. 

I think if you can get it working with nginx it should work with Apache as well, so I feel it's just a configuration issue. 

Something that wasn't included with the docs I looked at initially is that you need to enable the rewrite module as well. Here is that command
```
a2enmod rewrite
```

Also, for now, you would get the error popup but you aren't losing anything by not using the websocket because for what Overseer does ServerSentEvents work just as well. 

I probably need to handle that exception and not display the popup anyway because this is expected behavior when the connection to the websocket can't be made. 

It will try WebSockets, if that fails it will try SSE, if that fails it will just poll the server for updates. All of this should not be noticeable to the end user.

michaelfdeberry, 2019-12-17:
I installed Apache and was able to replicate the websocket issue. However, I am not getting any unknown error popup or messages in the log. The error regarding the websocket connection was logged but not triggering a popup, this is expected behavior. 

I was able to get the websocket working with Apache, here is the configuration I used.
```
<VirtualHost *:7500>
        RewriteEngine On
        RewriteCond %{HTTP:Upgrade} =websocket
        RewriteRule /(.*) ws://localhost:9000/$1 [P,QSA,L]
        RewriteCond %{HTTP:Upgrade} !=websocket
        RewriteRule /(.*) http://localhost:9000/$1 [P,QSA,L]

        ProxyPreserveHost On
        ProxyPass / http://localhost:9000/
        ProxyPassReverse / http://localhost:9000/
        ProxyRequests Off
</VirtualHost>
```

I am sure you reverse proxy configuration is fine, if you use this you likely only need the rewrite section from the top. Just make sure all the fore mentioned modules are enabled.

So, I am not sure what caused the unknown error to be logged. I have some client side errors in my log, but all of them have the correct message. 

Are you still getting this error?

taker218, 2019-12-17:
Thanks!
With this config it's all working now.

I still get the error in the console, but no popup message in the interface like it was before.

michaelfdeberry, 2019-12-17:
The pop-up had to be caused by something else. If it has resolved itself it could have been a cached script problem after the update, but please post if you see it again.

I'll leave this open for now.

michaelfdeberry, 2019-12-25:
I'm going to close this, if the error pops back up or you have any other problems please open a new issue.

closed 2019-12-25 by michaelfdeberry