OverseerApp/overseer #40: Can not add machines after update to current version.

Stady234, 2020-08-20:
Not sure what i am doing wrong... here is the log...
[overseer.log](https://github.com/michaelfdeberry/overseer/files/5100059/overseer.log)

michaelfdeberry, 2020-08-20:
I won't be able to really look into look into this until tomorrow evening, but after a quick glance at your log I see a NameResolutionFailure error.

This likely means Overseer can't find the Octoprint instance. Confirm that the ip address, or host name, is correct and that the device that is hosting Overseer can also access Octoprint.

Stady234, 2020-09-07:
ok, sorry for the long delay... work has been crazy... anyway i just redid the connections ad was able to add the latest 3 octoprint instances... no issues... idk what i did wrong... but looks like it was just me!

closed 2020-09-07 by Stady234