OverseerApp/overseer #90: Trouble with octoprint in subfolder

Benoit221, 2024-07-04:
Hello,

&ensp;I'm trying to configure a working standalone overseer before trying more difficult setup.

&ensp;So i configure my two octoprint in subfolder of my fqdn and overseer in root path of the same fqdn.

&ensp;Ender3 --> https://myfqdn/ender3
&ensp;Ender5 --> https://myfqdn/ender5
&ensp;Overseer --> https://myfqdn/

&ensp;But when i try to configure my ender3 instance, it refuse to save settings. In developers console i can see that it try to get https://myfqdn/api/settings instead of https://myfqdn/ender3/api/settings

&ensp;Is it possible to change this ?

Best regards,
B LORAND

michaelfdeberry, 2024-07-04:
Hi @Benoit221 ,

Are you using the client/server configuration of overseer? If so, I think `https://myfqdn/api/settings` may be the call to the overseer backend. 

I haven't looked at this code in a while, so I would need to review it to really remember how it all works. However, what error are you getting, and can you post the overseer logs?

Benoit221, 2024-07-05:
Hi @michaelfdeberry,

I'm using a selfhosted standalone using docker for now because i would like to check if i need to invistigate more this solution. As of i know https://myfqdn/api/settings should be given by octorpitn to validate settings and give information about connected printer. In fact, if i connect https://myfqdn/ender3/api/settings in my browser i can see a result.

I think overseer split url i put to only take https://myfqdn/ and not https://myfqdn/ender3.

here the log :

overseer  | /docker-entrypoint.sh: /docker-entrypoint.d/ is not empty, will attempt to perform configuration
overseer  | /docker-entrypoint.sh: Looking for shell scripts in /docker-entrypoint.d/
overseer  | /docker-entrypoint.sh: Launching /docker-entrypoint.d/10-listen-on-ipv6-by-default.sh
overseer  | 10-listen-on-ipv6-by-default.sh: info: Getting the checksum of /etc/nginx/conf.d/default.conf
overseer  | 10-listen-on-ipv6-by-default.sh: info: Enabled listen on IPv6 in /etc/nginx/conf.d/default.conf
overseer  | /docker-entrypoint.sh: Sourcing /docker-entrypoint.d/15-local-resolvers.envsh
overseer  | /docker-entrypoint.sh: Launching /docker-entrypoint.d/20-envsubst-on-templates.sh
overseer  | /docker-entrypoint.sh: Launching /docker-entrypoint.d/30-tune-worker-processes.sh
overseer  | /docker-entrypoint.sh: Configuration complete; ready for start up
overseer  | 2024/07/05 19:06:13 [notice] 1#1: using the "epoll" event method
overseer  | 2024/07/05 19:06:13 [notice] 1#1: nginx/1.26.1
overseer  | 2024/07/05 19:06:13 [notice] 1#1: built by gcc 12.2.0 (Debian 12.2.0-14)
overseer  | 2024/07/05 19:06:13 [notice] 1#1: OS: Linux 6.1.0-22-amd64
overseer  | 2024/07/05 19:06:13 [notice] 1#1: getrlimit(RLIMIT_NOFILE): 1048576:1048576
overseer  | 2024/07/05 19:06:13 [notice] 1#1: start worker processes
overseer  | 2024/07/05 19:06:13 [notice] 1#1: start worker process 28
overseer  | 2024/07/05 19:06:13 [notice] 1#1: start worker process 29
overseer  | 2024/07/05 19:06:24 [error] 29#29: *1 open() "/usr/share/nginx/html/assets/i18n/fr.json" failed (2: No such file or directory), client: 192.168.77.254, server: localhost, request: "GET /assets/i18n/fr.json HTTP/1.1", host: "myfqdn", referrer: "https://myfqdn/"
overseer  | 192.168.77.254 - - [05/Jul/2024:19:06:24 +0000] "GET /assets/i18n/fr.json HTTP/1.1" 404 555 "https://myfqdn/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" "192.168.77.102"
overseer  | 192.168.77.254 - - [05/Jul/2024:19:07:47 +0000] "GET /api/settings HTTP/1.1" 404 555 "https://myfqdn/configuration/machines/add" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" "192.168.77.102"
overseer  | 192.168.77.254 - - [05/Jul/2024:19:07:47 +0000] "GET /api/printerprofiles HTTP/1.1" 404 555 "https://myfqdn/configuration/machines/add" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" "192.168.77.102"
overseer  | 2024/07/05 19:07:47 [error] 28#28: *2 open() "/usr/share/nginx/html/api/settings" failed (2: No such file or directory), client: 192.168.77.254, server: localhost, request: "GET /api/settings HTTP/1.1", host: "myfqdn", referrer: "https://myfqdn/configuration/machines/add"
overseer  | 2024/07/05 19:07:47 [error] 29#29: *3 open() "/usr/share/nginx/html/api/printerprofiles" failed (2: No such file or directory), client: 192.168.77.254, server: localhost, request: "GET /api/printerprofiles HTTP/1.1", host: "myfqdn", referrer: "https://myfqdn/configuration/machines/add"

[image]

Best regards,
B LORAND

michaelfdeberry, 2024-07-05:
You are correct, the code expects octoprint at the root. 

Changing it so it could use a sub path is a minor change and wouldn't take long. However, this code is pretty old now so if I were to change anything I would also update the dependencies to the latest. That could take a while if there has been any breaking changes.

michaelfdeberry, 2024-08-04:
@Benoit221 it's been a little while so I wanted to give an update in case you are waiting on this. 

I have been working on Overseer when I have the time, but the code atrophied pretty bad. Lots of dependencies were deprecated or had major changes that required things to be reworked.

The good news is I now have it in a state where everything is updated and functional. However, the issue now is that Overseer uses the angular material library for UI. All the styling for Overseer targets the older version, so none of that works. It's using the default look and feel, and with that the UX is a little rough, which I feel effects the usability.

I could update the styling to improve things, but years back I had a designer mock up some designs that I never got around to implementing. If I have to update the styling, I would rather just implement those designs instead.

That said, the latest code is on the development branch. If you want to try it, it would require pulling the code and running the build yourself. If you are interested in that let me know and I will provide some instructions. Otherwise it would probably be a couple weeks before there is a release.

Benoit221, 2024-08-08:
Many thanks,I will check this this week-end i think.Best regards,Benoit

Benoit221, 2024-08-16:
Hi,

I just try to compile it using a Dockerfile like this :

```
FROM ubuntu:22.04 AS builder
LABEL Benoit LORAND <benoit.lorand@blorand.org>
ENV VERSION=1.0.23
ENV SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
ENV SSL_CERT_DIR=/dev/null

RUN apt-get update && apt-get install -y unzip dotnet8 ca-certificates npm && apt-get clean && update-ca-certificates

RUN mkdir -p /opt
WORKDIR /source
ADD https://github.com/michaelfdeberry/overseer/archive/refs/heads/development.zip /opt/development.zip
#ADD https://github.com/michaelfdeberry/overseer/archive/refs/tags/${VERSION}.zip /opt/${VERSION}.zip
RUN unzip /opt/development.zip
#RUN unzip /opt/${VERSION}.zip

# publish your .NET app
#WORKDIR /source/overseer-${VERSION}/src
WORKDIR /source/overseer-development/src

RUN dotnet publish -c Release -o /app

FROM ubuntu/dotnet-runtime:8.0-24.04_stable
WORKDIR /app
COPY --from=builder /app ./
ENTRYPOINT ["dotnet", "/app/Overseer.Daemon.dll"]
```

but it fail with :

```
------
 > [app builder 8/8] RUN dotnet publish -c Release -o /app:
2.224
2.240 Welcome to .NET 8.0!
2.240 ---------------------
2.240 SDK Version: 8.0.108
5.194
5.202 ----------------
5.202 Installed an ASP.NET Core HTTPS development certificate.
5.202 To trust the certificate, view the instructions: https://aka.ms/dotnet-https-linux
5.202
5.202 ----------------
5.202 Write your first app: https://aka.ms/dotnet-hello-world
5.202 Find out what's new: https://aka.ms/dotnet-whats-new
5.202 Explore documentation: https://aka.ms/dotnet-docs
5.202 Report issues and find source on GitHub: https://github.com/dotnet/core
5.202 Use 'dotnet --help' to see available commands or visit: https://aka.ms/dotnet-cli
5.202 --------------------------------------------------------------------------------------
6.084 MSBuild version 17.8.5+b5265ef37 for .NET
20.01   Determining projects to restore...
74.59   Restored /source/overseer-development/src/Overseer.Server/Overseer.Server.csproj (in 47.1 sec).
74.60   Restored /source/overseer-development/src/Overseer/Overseer.csproj (in 47.11 sec).
83.08   Restored /source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj (in 8.43 sec).
83.43 /usr/lib/dotnet/sdk/8.0.108/Current/SolutionFile/ImportAfter/Microsoft.NET.Sdk.Solution.targets(36,5): warning NETSDK1194: The "--output" option isn't supported when building a solution. Specifying a solution-level output path results in all projects copying outputs to the same directory, which can lead to inconsistent builds. [/source/overseer-development/src/Overseer.sln]
120.7   Overseer -> /source/overseer-development/src/Overseer/bin/Release/netstandard2.0/Overseer.dll
121.3   Overseer -> /app/
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103: Invalid Resx file. System.IO.DirectoryNotFoundException: Could not find a part of the path '/source/overseer-development/src/Overseer.Tests/resources/octoprintjob.txt'. [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at Interop.ThrowExceptionForIoErrno(ErrorInfo errorInfo, String path, Boolean isDirError) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, UnixFileMode openPermissions, Int64& fileLength, UnixFileMode& filePermissions, Boolean failForSymlink, Boolean& wasSymlink, Func`4 createOpenException) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at System.IO.Strategies.OSFileStreamStrategy..ctor(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize, Nullable`1 unixCreateMode) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, Int64 preallocationSize) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at System.IO.StreamReader.ValidateArgsAndOpenPath(String path, Encoding encoding, Int32 bufferSize) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at System.IO.StreamReader..ctor(String path, Encoding encoding) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
125.5 /source/overseer-development/src/Overseer.Tests/Properties/Resources.resx : error MSB3103:    at Microsoft.Build.Tasks.ResourceHandling.MSBuildResXReader.AddLinkedResource(String resxFilename, Boolean pathsRelativeToBasePath, List`1 resources, String name, String value) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
[... cut ...] Path, TaskLoggingHelper log, Boolean logWarningForBinaryFormatter) [/source/overseer-development/src/Overseer.Tests/Overseer.Tests.csproj]
166.6   Overseer.Server -> /source/overseer-development/src/Overseer.Server/bin/Release/net8.0/Overseer.Server.dll
167.3   Overseer.Server -> /app/
------
failed to solve: process "/bin/sh -c dotnet publish -c Release -o /app" did not complete successfully: exit code: 1
```

I haven't do any research on that but perhaps you already know what i'm doing wrong ?

michaelfdeberry, 2024-08-16:
If you are planning to use the .net build it's not ready yet. However, it looks like it's still looking for files that have been removed, I will have to fix that. The .Net build is also a little more involved because it would require building the front end in remote mode and copying the files to the build output of the .net app. 

However, I thought you were planning to use the standalone build. If that is the case you would only need to build the Overseer.Client directory, the commands are 

```
npm i 
npm run build
``` 

The content that would be served `dist\overseer.client\browser`. 

If you want to use the .Net build I can package up a pre-release version, I can make that available tomorrow.

Benoit221, 2024-08-16:
Many thanks ! i had successfully test it in dev mode. So i only see that webcam url is not good, because the subfolder appear twice (https://3d.blorand.org/ender3/ender3/webcam/?action=stream instead of https://3d.blorand.org/ender3/webcam/?action=stream) but it working after manually change it.

Now testing it good, yes i'm also interested by installing it as server. For sharing with friends. But you have time,

michaelfdeberry, 2024-08-17:
@Benoit221, I made some updates to the server version. I haven't tested this, but here are some updates to the configuration you posted. 

```
FROM ubuntu:22.04 AS builder
LABEL Benoit LORAND <benoit.lorand@blorand.org>
ENV VERSION=1.0.23
ENV SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
ENV SSL_CERT_DIR=/dev/null

RUN apt-get update && apt-get install -y unzip dotnet8 ca-certificates npm && apt-get clean && update-ca-certificates

RUN mkdir -p /opt
WORKDIR /source
ADD https://github.com/michaelfdeberry/overseer/archive/refs/heads/development.zip /opt/development.zip
RUN unzip /opt/development.zip
#RUN unzip /opt/${VERSION}.zip

# publish the UI, has to happen first since it clears the directory
WORKDIR /source/overseer-development/src/Overseer.Client
RUN npm i
RUN npx ng build --configuration production-remote --output-path /app 

# publish your .NET app
WORKDIR /source/overseer-development/src/Overseer.Server
RUN dotnet publish -c Release -o /app

FROM ubuntu/dotnet-runtime:8.0-24.04_stable
WORKDIR /app
COPY --from=builder /app ./
ENTRYPOINT ["dotnet", "/app/Overseer.Server.dll"]
```

You may also need to add nodejs to your apt-get list

Benoit221, 2024-08-18:
Thanks a lot @michaelfdeberry. It's working !

Some trouble persist, perhaps you are already knowing that, but in doubt :
[image]
[image]

Even if printing is running :
[image]

If i can help, just tell me.

Best regards,
B LORAND

michaelfdeberry, 2024-08-18:
I will take a look, but can you attach the log file and/or paste any errors from the browser console?

Benoit221, 2024-08-18:
Of course, here the browser console log :
```
ext::backendConnector: loaded namespace translation for language en Object
chunk-TZQWC3NE.js:2 i18next: languageChanged en
chunk-TZQWC3NE.js:2 i18next: initialized Object
main-Z5YNU4X7.js:8 [2024-08-18T11:40:37.492Z] Information: Normalizing '/push/status' to 'https://3d.blorand.org/push/status'.
main-Z5YNU4X7.js:8 [2024-08-18T11:40:37.556Z] Information: WebSocket connected to wss://3d.blorand.org/push/status?id=y7NkaSP5L0YmLuaZ4dKyFw.
6chunk-TZQWC3NE.js:2 i18next::translator: missingKey en translation unavailable 
```
And here the docker-compose log :
```
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
overseer  |       Overseer Scheme was not authenticated. Failure message: invalid_token
overseer  | info: Overseer.Server.OverseerAuthenticationHandler[7]
```

michaelfdeberry, 2024-08-19:
Nothing too specific in the logs, but the invalid_token could be a concern, I will see what I can figure out about that. Otherwise, it looks like it's making the connection to the websocket, so I am not sure what's going on there. 

If you filter the request to web sockets, and then look at the messages, is there anything coming or going. Here is an image if you aren't familiar: 
[image]

Are you using nginx? If so, you could take a look at this page https://github.com/michaelfdeberry/overseer/wiki/SSL-TLS-with-nginx. I suppose you could have similar issues with other reverse proxies, if you are using one.

michaelfdeberry, 2024-12-29:
@Benoit221, I know it's been a while, but I put out a new prerelease version. It should be somewhat more stable than the build you would have been using. I could also use some feedback if you are willing to give it a test.

Benoit221, 2025-01-02:
Hello @michaelfdeberry,

Sorry i don't have enough time last months to provide you debug messages. Yes i could take a look at your dev in the next weeks. Normaly it should be calmer at work.

Best regards,
BLORAND

Benoit221, 2025-01-11:
Hello @michaelfdeberry 

Good work, it's beautiful this new ui ! :)

I successfully build a new docker with this Dockerfile :
```
FROM ubuntu:24.04 AS builder
LABEL Benoit LORAND <benoit.lorand@blorand.org>
ENV VERSION=2.0.0-alpha.1
ENV SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
ENV SSL_CERT_DIR=/dev/null

RUN apt-get update && apt-get install -y unzip dotnet8 ca-certificates npm nodejs && apt-get clean && update-ca-certificates

RUN mkdir -p /opt
WORKDIR /source
#ADD https://github.com/michaelfdeberry/overseer/archive/refs/heads/development.zip /opt/development.zip
ADD https://github.com/michaelfdeberry/overseer/archive/refs/tags/${VERSION}.zip /opt/${VERSION}.zip
#RUN unzip /opt/development.zip
RUN unzip /opt/${VERSION}.zip

WORKDIR /source/overseer-${VERSION}/src/Overseer.Client
RUN npm i
RUN npx ng build --configuration production-remote --output-path /app

# publish your .NET app
#WORKDIR /source/overseer-${VERSION}/src
WORKDIR /source/overseer-${VERSION}/src/Overseer.Server
RUN dotnet build -c Release
RUN dotnet publish -c Release -o /app

FROM ubuntu/dotnet-aspnet:8.0-24.04_stable
WORKDIR /app
COPY --from=builder /app ./
ENTRYPOINT ["dotnet", "/app/Overseer.Server.dll"]
```

I also use this docker-compose.yml :
```
services:
  app:
    #    image: nginx:stable
    build: ./build
    restart: always
    container_name: overseer
    ports:
     - "8081:9000"
    volumes:
     - ./html:/usr/share/nginx/html

networks:
  default:
    ipam:
      config:
        - subnet: 172.18.1.0/24
```

It's seem that i don't found the file to be persistant for configuration to be saved.

Finally, all doesn't work as expected, see yourself with those screenshots :
[image]
Perhaps translation not found because my browser is setting to fr

[image]
My subfolder is setting twice for camera, so i need to adjust it.

[image]
Informations about printer are not displayed correctly.

Thanks for your work.
Best regards,
B LORAND

michaelfdeberry, 2025-01-11:
@Benoit221, Thanks for taking a look, your feedback is very helpful.

> Perhaps translation not found because my browser is setting to fr 

It looks like only the new strings aren't working. I am guessing the browser is using the language file from the cache. I should be able to add a cache busting query parameter. 

> My subfolder is setting twice for camera, so i need to adjust it. 

I must have broken that again. I will check it. 

> It's seem that i don't found the file to be persistant for configuration to be saved.

I think I know what is happening. The config db is in the app directory, and the directory will be cleared by this command `RUN npx ng build --configuration production-remote --output-path /app`. 

The fix for this would be to remove the `--output-path /app` argument and then copy the output from the dist directory into the app directory. I haven't tested the copy command, but it might look something like this: 

```
WORKDIR /source/overseer-${VERSION}/src/Overseer.Client
RUN npm i
RUN npx npm run build:remote 
COPY /dist/overseer.client.remote/browser /app 
```

> Informations about printer are not displayed correctly.

This one I am not sure about. Overseer will connect when saving the printer so it shouldn't have any issues connecting on the backend to get the updates. It has to be the websocket failing to connect. I haven't tested Overseer using docker, but I will try to do that this weekend.

michaelfdeberry, 2025-01-11:
The update issue isn't likely to be docker related. Using the configuration you posted above I can get status updates. 
 
If you don't mind, please provide the overseer logs, and if possible open the browsers dev tools and then refresh the page with the dev tools open. If there are any errors in the console please post them as well.

Benoit221, 2025-01-12:
You are right. After flushing browser cache. Text are displaying good.
It's working with a new virtual printer instance. So i update my old octoprint. And it seems to work. But i don't have any print to launch for now.

[image]

Benoit221, 2025-01-12:
Finally i think troubble appears when printer goes offline for a long time. When arrived, i can disable monitor and enable again and it works.

Benoit221, 2025-01-12:
Please find logfile here : [overseer.log](https://github.com/user-attachments/files/18390280/overseer.log)

I don't see anything bad in the developpers tools.

Best regards,
B LORAND

michaelfdeberry, 2025-01-12:
I think I have a better idea of what is going on. You have a few 409 responses from Octoprint in the log. That is a documented response [`409 Conflict – If the printer is not operational.`](https://docs.octoprint.org/en/master/api/printer.html#retrieve-the-current-printer-state). It likely means the printer is disconnected in Octoprint. 

Overseer polls at a reduced rate after a certain number of consecutive failures and won't start polling regularly again until it gets a successful response. So, that is likely why it seems like it takes a while for it to comeback after it's connected. I also think disabling and then enabling monitoring would reset the timeout period. 

It would be possible for Overseer to use the API to connect the printer. Currently it intentionally doesn't do that, because my thought there is if it's disconnected in Octoprint it's likely intentional. However, that may make a good optional feature. 

There is also this error `The input string 'W' was not in a correct format.` Given where this is failing it would mean Octoprint is returning something that isn't documented. I think the best I can do about that for now is to add some additional logging.

michaelfdeberry, 2025-01-26:
I think the duplicate segment issue in the webcam url should be resolved in the latest version. I have created #111 for the optional feature of having Overseer attempt to get Octoprint to open a connection to the printer. 

I am going to close this issue. If you notice any other bugs feel free to reopen or create a new issue if it's something completely different. Thank you for testing the new version.

Benoit221, 2025-01-26:
Hello Michael,

Effectively, Camera configuration is good now.

I always have trouble with offline printer with alpha6.

I found that some api access (/api/settings and /api/machines) are arriving to overseer instead of octoprint. Could you check if the correct suburl is use for this ?

Best regards,
B LORAND

michaelfdeberry, 2025-01-27:
> Hello Michael,
>
> Effectively, Camera configuration is good now.
>
> I always have trouble with offline printer with alpha6.
>
> I found that some api access (/api/settings and /api/machines) are arriving [...]

You are correct, it looks like I caused a regression with the camera path fix. I will revert the last change and try something different.

michaelfdeberry, 2025-01-27:
Looking at your comment again, I think those are request to the Overseer API. 

That is if you are still using the version of Overseer with a backend. If you are seeing the request in the browser dev tools, then that is the case.  In backend version of Overseer the request to Octoprint are made from the server. This way, if multiple clients are connected, it can send a single update to all clients, instead of having all client poll Octoprint individually. 

Are the request listed failing? If so please post the error messages.

Benoit221, 2025-01-27:
You are rigth, it's overseer API. I still use it with the backend. Those request get a 200 code so it's not that.

I check overseer.log and i get this many times :
```
2025-01-27 19:27:48,591 [14] ERROR Overseer.Machines.MachineProvider`1[[Overseer.Models.OctoprintMachine, Overseer, Version=2.0.0.0, Culture=neutral, PublicKeyToken=null]] [(null)] - Max consecutive failure count reached, throttling updates
System.Text.Json.JsonException: The JSON value could not be converted to System.Double. Path: $.temperature.W.target | LineNumber: 25 | BytePositionInLine: 20.
 ---> System.InvalidOperationException: Cannot get the value of a token type 'Null' as a number.
   at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_ExpectedNumber(JsonTokenType tokenType)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonDictionaryConverter`3.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, TDictionary& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, JsonSerializerOptions options, ReadStack& state)
   --- End of inner exception stack trace ---
   at Overseer.Machines.PollingMachineProvider`1.Fetch[T](String resource, FetchRequest fetchRequest, CancellationToken cancellation) in /source/overseer-2.0.0-alpha.6/src/Overseer/Machines/PollingMachineProvider.cs:line 156
   at Overseer.Machines.Octoprint.OctoprintMachineProvider.AcquireStatus(CancellationToken cancellationToken) in /source/overseer-2.0.0-alpha.6/src/Overseer/Machines/Octoprint/OctoprintMachineProvider.cs:line 156
   at Overseer.Machines.PollingMachineProvider`1.Poll() in /source/overseer-2.0.0-alpha.6/src/Overseer/Machines/PollingMachineProvider.cs:line 83
2025-01-27 19:27:58,590 [58] ERROR Overseer.Machines.MachineProvider`1[[Overseer.Models.OctoprintMachine, Overseer, Version=2.0.0.0, Culture=neutral, PublicKeyToken=null]] [(null)] - Max consecutive failure count reached, throttling updates
System.Text.Json.JsonException: The JSON value could not be converted to System.Double. Path: $.temperature.W.target | LineNumber: 25 | BytePositionInLine: 20.
 ---> System.InvalidOperationException: Cannot get the value of a token type 'Null' as a number.
   at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_ExpectedNumber(JsonTokenType tokenType)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonDictionaryConverter`3.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, TDictionary& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value, Boolean& isPopulatedValue)
   at System.Text.Json.Serialization.JsonConverter`1.ReadCore(Utf8JsonReader& reader, JsonSerializerOptions options, ReadStack& state)
   --- End of inner exception stack trace ---
   at Overseer.Machines.PollingMachineProvider`1.Fetch[T](String resource, FetchRequest fetchRequest, CancellationToken cancellation) in /source/overseer-2.0.0-alpha.6/src/Overseer/Machines/PollingMachineProvider.cs:line 156
[... cut ...]
   at Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddlewareImpl.<Invoke>g__Awaited|10_0(ExceptionHandlerMiddlewareImpl middleware, HttpContext context, Task task)
2025-01-27 19:36:46,532 [79] ERROR Overseer.Server.Api.LoggingApi [(null)] - 01/27/2025 19:36:46 - ERROR in : Failed to establish a connection with the machine
```

But if i delete My Ender5 machine, i can add it again. But even if i delete Ender5 machine, i can't add my virtual printer who was working before.

Best regards,
B LORAND

Benoit221, 2025-01-27:
I just test to delete my docker and rerun it.
For first printer i declare the virtual who was accepted.
As second i declare my ender5 and get the error : "Failed to establish a connection with the machine"

michaelfdeberry, 2025-01-28:
It looks like I missed that the target temp can be null. I think I have only ever gotten 0 back with my testing. Fixing that should resolve the update issue. 

This is the error that is being thrown when trying to add a new printer `Request failed with status code Forbidden`, I think either the API key is incorrect or it was create for a user that doesn't have the READ permission.

michaelfdeberry, 2025-04-11:
I did end up finding a bug that I think was causing the issue you described. That would have been fixed in one of the newer versions. This issue is getting kind of big, so if you have any additional problems please create a new issue.

closed 2025-04-11 by michaelfdeberry